MagicMirrorOrg/MagicMirror #2838: MagicMirror2 not starting properly

notoriousNichie, 2022-04-14:
I keep receiving a screen saying " MagicMirror2... please create a config file or check the existing one for errors" when I try to run my magicmirror2.

[https://www.dropbox.com/s/vv5yr7jllnpotpn/Photo%20Apr%2013%2C%204%2049%2003%20PM.jpg?dl=0](url) 

When I run the command npm run config:check, I receive no errors.  How can I fix this?

my config file:

/* MagicMirror\u00b2 Config Sample 
let config = {
    address: "localhost", // Address to listen on, can be:
    // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
    // - another specific IPv4/6 to listen on a specific interface
    // - "0.0.0.0", "::" to listen on any interface
    // Default, when address config is left out or empty, is "localhost"
    port: 8080,
    basePath: "/", // The URL path where MagicMirror\u00b2 is hosted. If you are using a Reverse proxy
    // you must set the sub path here. basePath must end with a /
    ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
    // or add a specific IPv4 of 192.168.1.5 :
    // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
    // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
    // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

    useHttps: false, // Support HTTPS or not, default "false" will use HTTP
    httpsPrivateKey: "", // HTTPS private key path, only require when useHttps is true
    httpsCertificate: "", // HTTPS Certificate path, only require when useHttps is true

    language: "en",
    locale: "en-US",
    logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
    timeFormat: 24,
    units: "metric",
    // serverOnly:  true/false/"local" ,
    // local for armv6l processors, default
    //   starts serveronly and then starts chrome browser
    // false, default for all NON-armv6l devices
    // true, force serveronly mode, because you want to.. no UI on this device

    modules: [
        {
            module: "alert",
        },
        {
            module: "updatenotification",
            position: "top_bar"
        },
        {
            module: "clock",
            position: "top_left"
        },
        {
            module: "calendar",
            header: "US Holidays",
            position: "top_left",
            config: {
                calendars: [{
                    symbol: "calendar-check",
                    url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics [calendarlabs.com]"
                }]
            }
        },
        {
            module: "compliments",
            position: "lower_third"
        },
        {
            module: 'MMM-horoscope',
            position: 'middle_center', // This can be any of the regions.
            config: {
                // See 'Configuration options' for additional options and more information.
                sign: 'aquarius', // Zodiac sign
                maxWidth: "400px", // maximum width of the module in px, %, em
                updateInterval: 1 * 60 * 60 * 1000, // updates every hour
                useTextIcon: true,
                initialLoadDelay: 0,
                animationSpeed: 2000,
            }
        },
        {
            module: "MMM-Wallpaper",
            position: "fullscreen_below",
            config: { // See "Configuration options" for more information.
                source: "icloud:B0p5UzI7VEutng",
                slideInterval: 60 * 1000 // Change slides every minute
            }
        },
        {
            module: "weather",
            position: "top_right",
            header: "Weather Forecast",
            config: {
                weatherProvider: "openweathermap",
                type: "forecast",
                location: "Birmingham",
                locationID: "4049979", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz [bulk.openweathermap.org]; unzip the gz file and find your city
                apiKey: "https://api.openweathermap.org/data/2.5/weather?zip=35064,us&appid=0902e3c19914fa6022959c47ccf48dab [api.openweathermap.org]"
            }
        },
        {
            module: "newsfeed",
            position: "bottom_bar",
            config: {
                feeds: [{
                    title: "New York Times",
                    url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml [rss.nytimes.com]"
                }],
                showSourceTitle: true,
                showPublishDate: true,
                broadcastNewsFeeds: true,
                broadcastNewsUpdates: true
            }
        },
    ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

sdetweil, 2022-04-14:
look at the rest of the messages where you do npm start..

if using pm2 then use the command 
 pm2 logs --lines=50

notoriousNichie, 2022-04-14:
> look at the rest of the messages where you do npm start..
> 
> if using pm2 then use the command pm2 logs --lines=50

I do not understand what you mean by this? can you explain?

sdetweil, 2022-04-14:
I need to see the rest of the messages output by MagicMirror starting up

u either issued a command. 
npm start 
in the MagicMirror folder
or
you use pm2 to start MagicMirror
if pm2, then u can get the messages by using the command
pm2 logs --lines=50   
(or some other number than 50)

notoriousNichie, 2022-04-15:
pi@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.19.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[15.04.2022 18:06.32.916] [LOG]   Starting MagicMirror: v2.19.0
[15.04.2022 18:06.32.924] [LOG]   Loading config ...
[15.04.2022 18:06.32.931] [ERROR] WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: ReferenceError: config is not defined
    at Object.<anonymous> (/home/pi/MagicMirror/config/config.js:110:54)
    at Module._compile (node:internal/modules/cjs/loader:1116:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at loadConfig (/home/pi/MagicMirror/js/app.js:69:14)
    at App.start (/home/pi/MagicMirror/js/app.js:212:3)
[15.04.2022 18:06.32.933] [LOG]   Loading module helpers ...
[15.04.2022 18:06.32.969] [LOG]   Initializing new module helper ...
[15.04.2022 18:06.32.972] [LOG]   Module helper loaded: updatenotification
[15.04.2022 18:06.32.976] [LOG]   No helper found for module: helloworld.
[15.04.2022 18:06.32.978] [LOG]   All module helpers loaded.
[15.04.2022 18:06.33.177] [LOG]   Starting server on port 8080 ... 
[15.04.2022 18:06.33.203] [LOG]   Server started ...
[15.04.2022 18:06.33.204] [LOG]   Connecting socket for: updatenotification
[15.04.2022 18:06.33.205] [LOG]   Starting module helper: updatenotification
[15.04.2022 18:06.33.206] [LOG]   Sockets connected & modules started ...
[15.04.2022 18:06.33.824] [LOG]   Launching application.
[15.04.2022 18:06.35.686] [INFO]  Checking git for module: default
[15.04.2022 18:06.39.979] [LOG]   Shutting down server...
[15.04.2022 18:06.39.982] [LOG]   Stopping module helper: updatenotification

notoriousNichie, 2022-04-15:
that is what i reeive when i do a NPM run start on my magic mirror

sdetweil, 2022-04-15:
how did you install MagicMirror?

if you did the git clone/npm install
then you need to copy the config.js.sample to config.js 
before starting the first time

step 5 of the instructions
[image]

notoriousNichie, 2022-04-16:
I made a clean install, followed the steps in order, and when i run npm run start ( the mirror works normally). Then After i install the modules and add the configs to the config.js. i get errors when i run "npm run start".. looks like the following :
 

pi@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.19.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[16.04.2022 18:17.43.267] [LOG]   Starting MagicMirror: v2.19.0
[16.04.2022 18:17.43.275] [LOG]   Loading config ...
[16.04.2022 18:17.43.281] [ERROR] WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: /home/pi/MagicMirror/config/config.js:86
		{
		^

SyntaxError: Unexpected token '{'
    at Object.compileFunction (node:vm:352:18)
    at wrapSafe (node:internal/modules/cjs/loader:1038:15)
    at Module._compile (node:internal/modules/cjs/loader:1072:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at loadConfig (/home/pi/MagicMirror/js/app.js:69:14)
[16.04.2022 18:17.43.283] [LOG]   Loading module helpers ...
[16.04.2022 18:17.43.301] [LOG]   Initializing new module helper ...
[16.04.2022 18:17.43.303] [LOG]   Module helper loaded: updatenotification
[16.04.2022 18:17.43.306] [LOG]   No helper found for module: helloworld.
[16.04.2022 18:17.43.308] [LOG]   All module helpers loaded.
[16.04.2022 18:17.43.492] [LOG]   Starting server on port 8080 ... 
[16.04.2022 18:17.43.514] [LOG]   Server started ...
[16.04.2022 18:17.43.516] [LOG]   Connecting socket for: updatenotification
[16.04.2022 18:17.43.517] [LOG]   Starting module helper: updatenotification
[16.04.2022 18:17.43.517] [LOG]   Sockets connected & modules started ...
[16.04.2022 18:17.43.928] [LOG]   Launching application.
[16.04.2022 18:17.45.775] [INFO]  Checking git for module: default
[5093:0416/181815.973484:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!

sdetweil, 2022-04-16:
yes you have a typographical error

unexpected token almost always is a missing trailing comma on the line BEFORE the line the error is reported on

in your case, on line 85

notoriousNichie, 2022-04-16:
can you tell me what typographical error i need to change:

/* MagicMirror\u00b2 Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror\u00b2 is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"
					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
 			module: 'MMM-PokemonGOEvents',
			position: 'bottom_right',
			header: "Pokemon GO Events",
			config: {
		        category: "current",
		        theme: "default",
		        updateInterval: 5000,
		        maxEvents: 5,
		        truncateTitle: 0,
		        exactTimestamp: false,
		        eventWhitelist: [],
		        eventBlacklist: [],
		        specificEventBlacklist: [],
		        eventIcon: "fa-solid fa-ticket"
		    }
		}
		{
			module: 'MMM-iCloud-Client',
			config: {
				username: "drea199115@yahoo.com",
				password: "Nichie22@",
				countrycode: "+1",
		}
		},
		{
			symbol: 'birthday-cake',
			url: 'http://localhost:8080/MMM-iCloud-Client/birthdays',
			}

		{
			
 		         module: 'MMM-horoscope',
		         position: 'middle_center', // This can be any of the regions.
		            config: {
                // See 'Configuration options' for additional options and more information.
                	 sign: 'aquarius', // Zodiac sign
	                 maxWidth: "400px", // maximum width of the module in px, %, em
         	         updateInterval: 1 * 60 * 60 * 1000, // updates every hour
	                 useTextIcon: true,
         	         initialLoadDelay: 0,
	                 animationSpeed: 2000,
         	  }
        },
        {
            module: "MMM-Wallpaper",
            position: "fullscreen_below",
            config: { // See "Configuration options" for more information.
                source: "icloud:B0p5UzI7VEutng",
                slideInterval: 60 * 1000 // Change slides every minute
            }
        },
        {
            module: "weather",
            position: "top_right",
            header: "Weather Forecast",
            config: {
                weatherProvider: "openweathermap",
                type: "forecast",
                location: "Birmingham",
                locationID: "4049979", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
                apiKey: "https://api.openweathermap.org/data/2.5/weather?zip=35064,us&appid=0902e3c19914fa6022959c47ccf48dab"
            }
        },
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

sdetweil, 2022-04-16:
yes, but I want you to find it

every module is
```
{
   module: name,
    ...
}
```

and if there is ANOTHER module following, this module definition needs a , after the }

means more follows
u left the comma ','
off

sdetweil, 2022-09-13:
we can close this

closed 2022-09-13 by MichMich