MagicMirrorOrg/MagicMirror #2840: Impossible to display full article

josselynL, 2022-04-22:
Hello everyone,

I want to see details of current news displayed with notification ARTICLE_MORE_DETAIL or ARTICLE_TOGGLE_FULL notification, but this is not functionnal (just display black screen)

Logs
```
0 | mm
(node:26811) electron: Failed to load URL: https://XXX  with error:ERR_BLOCKED_BY_RESPONSE
```

I think this is because the website block iframe to avoid click-jacking (https://stackoverflow.com/questions/48668767/error-when-loading-an-external-url-in-an-iframe-with-electron), so there is no possibilities in this case ?

Thank you for your help !

khassel, 2022-10-02:
This should be possible since https://github.com/MichMich/MagicMirror/pull/2850

Can this be closed? @josselynL

josselynL, 2022-10-03:
> This should be possible since #2850
> 
> Can this be closed? @josselynL

Thank you for this information.
I will try and come back to you soon with the result

josselynL, 2022-10-07:
@khassel 
MM freshly updated, but problem is still the same with same error.
Does I need to do something ?

Thank you.

khassel, 2022-10-07:
> Does I need to do something ?

you have to add the line `httpHeaders` into your `config.js` as provided below:

```js
let config = {
	address: "localhost",
	port: 8080,
	httpHeaders: { contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false, frameguard: false },
	...
}
```

josselynL, 2022-10-07:
> ```js
> 	httpHeaders: { contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false, frameguard: false },
> ```

Thank you for your replay.
But still the same issue.

`0|mm  | (node:27973) electron: Failed to load URL: https://actu.fr/occitanie/toulouse_31555/***********.html with error: ERR_BLOCKED_BY_RESPONSE
`

khassel, 2022-10-07:
can you provide your `config.js` with the related module config so we can reproduce this error?

Please remove all private information as e.g. apikeys before ...

josselynL, 2022-10-07:
Thank you.
Here is. I've delete all modules configuration, I think you don't need it :)

`js
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/getting-started/configuration.html#general
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "0.0.0.0", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	httpHeaders: { contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false, frameguard: false },
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: [], // Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "fr",
	locale: "fr",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
      		module: 'MMM-ModuleScheduler',
  		},
		{
            module: 'MMM-Carousel',
		},
		{
		module: 'MMM-Remote-Control',
		// uncomment the following line to show the URL of the remote control on the mirror
		// position: 'bottom_left',
		// you can hide this module afterwards from the remote control itself
    		config: {
        		customCommand: {},  // Optional, See "Using Custom Commands" below
        		showModuleApiMenu: true, // Optional, Enable the Module Controls menu
        		secureEndpoints: true, // Optional, See API/README.md
        		// uncomment any of the lines below if you're gonna use it
        		// customMenu: "custom_menu.json", // Optional, See "Custom Menu Items" below
        		apiKey: "***", // Optional, See API/README.md for details
        		// classes: {} // Optional, See "Custom Classes" below
					}
		},
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_center",
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
		},
		{
			module: "weather",
		},
		{
			module: "weather",
		},
		{
			module: "MMM-VigilanceMeteoFrance",
		},
        {
            module: 'compliments',
            //position: 'lower_third',
            position: 'middle_center',
				config: {
					morningStartTime:5,
					morningEndTime:12,
					afternoonStartTime:12,
					afternoonEndTime:19,
					remoteFile: 'compliments.json'
						}
        },
        {
			module: 'MMM-Jeedom',
			
        },
        {
			module: 'MMM-Jeedom',
        },
		{
            module: 'MMM-Jeedom',
		},
		{
			module: 'MMM-Jeedom',
        },
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
				/*	{
						title: "La depeche",
						url: "https://www.ladepeche.fr/rss.xml"
					},*/
					{
						title: "Actu Toulouse",
						url: "https://actu.fr/toulouse/rss.xml"
					},
					{
						title: "Netflix News",
						url: "https://www.netflix-news.com/feed/"
					},
					{
						title: "Cookomix : Les dernières recettes publiées",
						url: "https://cookomix.com/feed/"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true,
				ignoreOldItems: true,
			}
		},
		{
            module: 'MMM-Jeedom',
        },
		{
			module: 'MMM-Jeedom',
		},
        {
            module: 'EXT-Screen',
        },
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
`

khassel, 2022-10-07:
you can remove the `httpHeaders` stuff from above, this is for another usecase (embedding mm in another website).

As a first hack you can edit `modules/default/newsfeed/newsfeed.js` at about line nr. 150 and replace

```js
	getActiveItemURL: function () {
		return typeof this.newsItems[this.activeItem].url === "string" ? this.newsItems[this.activeItem].url : this.newsItems[this.activeItem].url.href;
	},
```

with

```js
	getActiveItemURL: function () {
		const urlPrefix = location.protocol + "//" + location.host + "/cors?url=";
		return typeof this.newsItems[this.activeItem].url === "string" ? urlPrefix + this.newsItems[this.activeItem].url : urlPrefix + this.newsItems[this.activeItem].url.href;
	},
```

Let me know if this works.

josselynL, 2022-10-08:
> 

Hi,
Yeah, this works better now ! Thank you 👍 
ARTICLE_MORE_DETAILS or ARTICLE_TOGGLE_FULL have the same result.
ARTICLE_MORE_DETAILS again and the page scroll down, but I can't go to the end of the page. It is blocking before, but I think this is another issue.

pm2 logs when I request ARTICLE_MORE_DETAILS for the first time
```
0|mm  | [08.10.2022 12:07.39.753] [LOG]
0|mm  | cors url: https://actu.fr/occitanie/toulouse_31555/bal-concerts-gratuits-expo-vide-dressing-10-sorties-a-faire-a-toulouse-ce-week-end_54337049.html
0|mm  | [08.10.2022 12:07.40.660] [LOG]
0|mm  | cors url: https://actu.fr/occitanie/toulouse_31555/top-14-un-stade-toulousain-en-grande-forme-recoit-clermont-pas-en-reussite-a-l-exterieur_54350507.html
0|mm  | [08.10.2022 12:07.41.208] [ERROR]
0|mm  | (node:28856) electron: Failed to load URL: https://actu.fr/checksso.html#error=login_required&state=d54f557c-e474-441c-8e02-eac37782bf62 with error: ERR_BLOCKED_BY_RESPONSE
```

The displayed article is the second URL, I don't know why the URL of previous article is in log.

rejas, 2022-10-08:
So, this issued can be closed now?

sdetweil, 2022-10-08:
can't scroll down is probably the main.css overflow:hidden

khassel, 2022-10-08:
> So, this issued can be closed now?

not yet, have to create a PR for this ...

josselynL, 2022-10-09:
Hello

@khassel TY for the PR
@sdetweil How can I check this ?

sdetweil, 2022-10-09:
@josselynL  find a big article and cause that to be displayed. did it get clipped at the bottom?  
will u have a mouse?  if so, scroll down with overflow:hidden turned off(commented out)

josselynL, 2022-10-09:
> @josselynL find a big article and cause that to be displayed. did it get clipped at the bottom? will u have a mouse? if so, scroll down with overflow:hidden turned off(commented out)

Just tried with Remote Control (and so a mouse).
When the scroll down is blocked with "ARTICLE_MORE_DETAILS", I can continue to scroll down with the mouse, until the end of the page.
Comment or uncomment the overflow has no impact.
```

html {
  cursor: none;
/*  overflow: hidden;*/
```

khassel, 2023-01-03:
the main issue is solved with release `v2.22.0`.

> ARTICLE_MORE_DETAILS again and the page scroll down, but I can't go to the end of the page. It is blocking before, but I think this is another issue.

If this is still a problem please open a new issue, thanks.

closed 2023-01-03 by khassel