MagicMirrorOrg/MagicMirror #2845: Authorization Error

01Anand, 2022-05-03:
When I configure Fitness API I getting following error ,

[03.05.2022 17:39.32.639] [ERROR] ReferenceError: tmpAuthData is not defined
    at Request._callback (/media/sf_projects/smartmirror/MagicMirror-master/modules/MMM-GoogleFit/node_helper.js:49:29)
    at Request.self.callback (/media/sf_projects/smartmirror/MagicMirror-master/node_modules/request/request.js:185:22)
    at Request.emit (events.js:198:13)
    at Request.<anonymous> (/media/sf_projects/smartmirror/MagicMirror-master/node_modules/request/request.js:1154:10)
    at Request.emit (events.js:198:13)
    at IncomingMessage.<anonymous> (/media/sf_projects/smartmirror/MagicMirror-master/node_modules/request/request.js:1076:12)
    at Object.onceWrapper (events.js:286:20)
    at IncomingMessage.emit (events.js:203:15)
    at endReadableNT (_stream_readable.js:1145:12)
    at process._tickCallback (internal/process/next_tick.js:63:19)
[03.05.2022 17:39.32.639] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[03.05.2022 17:39.32.640] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
^C[03.05.2022 17:53.02.834] [LOG]   [SIGINT] Received. Shutting down server...
[03.05.2022 17:53.02.834] [LOG]   Stopping module helper: MMM-GoogleFit

rejas, 2022-05-03:
You should ask this either at the modules repo (https://github.com/amcolash/MMM-GoogleFit) or in the forums (https://forum.magicmirror.builders/)

Since the module is abandoned, the later should be your way to go.

khassel, 2022-10-02:
no update since May, should be closed @01Anand @MichMich

closed 2022-10-03 by MichMich