MagicMirrorOrg/MagicMirror #2846: Issue with config lines

mblack1986, 2022-05-07:
Hi, I have been trying for a while to sort this out but to no avail. My MM was working great before I added the Google maps module. Now every time I run the npm config:check I get this error - 

[07.05.2022 13:03.32.086] [INFO]  Checking file...  /home/pi/MagicMirror/config/config.js
[07.05.2022 13:03.32.226] [ERROR] Your configuration file contains syntax errors :(
[07.05.2022 13:03.32.228] [ERROR] Line 201 column 2: Parsing error: Unexpected token ;

I have checked the forum and have tried to amend the } but this has not worked. Can anyone see any issues with my code? I am new to this so I am most certain I have done something wrong.

Thanks in advance.

Matt
(I have removed personal info)
-------------------------------

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: [], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-GB",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
		module: 'MMM-Todoist',
		position: 'top_center',	// This can be any of the regions. Best results in left or right regions.
		header: 'To-Do List', // This is optional
		config: { // See 'Configuration options' for more information.
			hideWhenEmpty: false,
			accessToken: '*****',
			maximumEntries: 60,
			updateInterval: 1*2*1000, // Update every 10 minutes
			fade: false,      
			// projects and/or labels is mandatory:
			projects: [ ***, ***, ***, ***, ***, *** ], 
			labels: [ "MagicMirror", "Important" ] // Tasks for any projects with these labels will be shown.
		}
		},
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_bar"
		},
		
		{
			module: "calendar",
			header: "Family Calendar",
			position: "top_left",
			config: {
					fade : true,
					fadePoint : 0.5,
					timeFormat : "absolute",
					dateFormat : "ddd Do MMMM",
					dateEndFormat : "HH:mm",
					urgency : "1",
					getRelative : "12",
					maximumEntries: 8,
				calendars: [
					{
						symbol: "Matt",
						url: "webcal://***"					
					},
					{
						symbol: "UK Holidays",
						url: "https://www.calendarlabs.com/ical-calendar/ics/75/UK_Holidays.ics"					
					},
					{
						symbol: "F1 2021",
						url: "webcal://ics.fixtur.es/v2/league/formula-1-qualifiers.ics"					
					},
					{
						symbol: "Family",
						url: "webcal://***"					
					}
				]
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Today's Weather",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "***",
				locationID: "***", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "***"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "***",
				locationID: "***", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "***"
			}
		},
		{
            module: 'MMM-GoogleMapsTraffic',
            position: 'bottom_right',
            config: {
                key: '***',
                lat: ***,
                lng: ***,
                height: '300px',
                width: '300px',
                disableDefaultUI: true,
                backgroundColor: 'hsla(0, 0%, 0%, 0)',
                markers: [
                    {
                        lat: ***,
                        lng: ***,
                        fillColor: '#9966ff' 
             },
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				wrapTitle: false,
				truncDescription: true,
				lengthDescription: 150,
				feeds: [
					{
						title: "Independent",
						url: "http://www.independent.co.uk/news/uk/rss"
					},
					{
						title: "ESPN F1",
						url: "http://en.espn.co.uk/rss/sport/story/feeds/0.xml?sport=4;type=2"
					},
					{
						title: "Autosport",
						url: "https://www.autosport.com/rss/f1/news/"
					},
					{
						title: "Guardian F1",
						url: "https://www.theguardian.com/sport/formulaone/rss"
					},
					{
						title: "Independent F1",
						url: "https://www.independent.co.uk/sport/motor-racing/formula1/rss"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
		{
			module: "MMM-Ring",
			position: "middle_center",
			config: {
				ring2faRefreshToken: "***",
				ringMinutesToStreamVideo: 1,
			}
		},
		{
		  module: 'MMM-MyGarbage',
		  position: 'top_left',
		  header: 'Bin Collections',
		  config: {
		          alert: 4,
		          weeksToDisplay: 2,
		          limitTo: 99,
		          fade: true,
		          dateFormat: "dddd Do MMMM",
		          fadePoint: 0.25
			}
		}
	],
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

KristjanESPERANTO, 2022-05-07:
The config of MMM-GoogleMapsTraffic is the problem. The markers part is missing closing brackets.

```js
    {
      module: 'MMM-GoogleMapsTraffic',
      position: 'bottom_right',
      config: {
        key: '***',
        lat: ** * ,
        lng: ** * ,
        height: '300px',
        width: '300px',
        disableDefaultUI: true,
        backgroundColor: 'hsla(0, 0%, 0%, 0)',
        markers: [{
          lat: ** * ,
          lng: ** * ,
          fillColor: '#9966ff'
        }]
      }
    },
```

mblack1986, 2022-05-07:
Ah that's excellent! Thank you so much! 

To help me learn and understand; why do these lines (or is it a string?) need closing brackets? and what are their purpose? also why did they config:check not pick this up but instead gave me an error at the end of the code (is that the right word?)

sdetweil, 2022-05-07:
@mblack1986 

why?

because there were starting opens   
```
        markers: [{    //  <---   these are the opens 
          lat: ** * ,
          lng: ** * ,
          fillColor: '#9966ff'
        }]   // <---  these are the missing closes 
```

the compiler cannot tell  WHERE they are 'supposed' to be..   only that it got to something that should NOT be here   
 Line 201 column 2: Parsing error: Unexpected token ;  

this error "Unexpected token"  means something is missing, cause the compiler is confused.. 

to close the [{   first u need a }

well that was here 
```
         },    <----
	{
		module: "newsfeed",
```
and a comma (more to follow) and then another { 
arrays []  of structures {} are normal in javascript.. so  no error yet...
and everything is valid til the ;...  (not inside an array for sure)

mblack1986, 2022-05-07:
@sdetweil amazingly explained, thank you very much! 

Sorry for further questions, last one I hope lol does it need to even have the [] within the markers string?

sdetweil, 2022-05-07:
@mblack1986 yes, markers is a list

in JavaScript a list is [ ]

with things inside it
numbers
```
  [ 1,2,3,89,0]
```
strings
```
[ "hello", "Bob", "world"]
```
and objects
```
[   { test:1, item:2, music:{ beathoven:"opera"}}, { something:"else"}}  ]
```
note that first object contains an object

mblack1986, 2022-05-08:
Thank you for taking the time to explain all of this to me, it has really made a difference! take care

closed 2022-05-08 by mblack1986

sdetweil, 2023-11-26:
@Bu61 thank you for the nice feedback.  if we all help others, the world will be a better place.