MagicMirrorOrg/MagicMirror #2847: Embedding in another Website

CodingCody-1, 2022-05-09:
I would like to embed MM in another website but every time I try it is just a white rectangle. From what I can tell it it because MM needs to allow it at some point.

I'm on the develop branch and have node version 16.14.2

sdetweil, 2022-05-09:
embed?  

are u running server mode?
what is your mm address/ipWhitelist settings?

where is the website, outside the network where mm is located?

how are u embedding? iframe?

CodingCody-1, 2022-05-11:
Yes, I'm running in server mode. It's all local only. Just typing in the IP and Port 8087. I don't have anything in the IP whitelist. I'm trying to embed it with iframe in Grafana.

sdetweil, 2022-05-11:
8087 is the configured mm port? in config.js?
if u open a browser, then mm comes up ok using that url. http://IP:port

sdetweil, 2022-05-11:
if I create a simple web page to use iframe
```html
<html>
<head>
</head>
<body>
	<iframe src="http://localhost:8090"></iframe>
</body>
</html>
```
I get an error in the dev window console 
```
Refused to display 'http://localhost:8090/' in a frame because it set 'X-Frame-Options' to 'sameorigin'.
```

but the code explicitly turns off this restriction  (in js/server.js, line 72) 
```javascript
app.use(helmet({ contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false }));
```

@khassel   you have any idea about this.. you added the cors redirector

khassel, 2022-05-11:
managed getting this to work by adding the frameguard option, see below git diff.

This will decrease security again but the whole app is so insecure ... I can provide a PR if wished, may with a config option so this is not allowed by default.

```js
magic_mirror on  develop [✘!?] is 📦 v2.20.0-develop via  v18.1.0 took 6s
⬢ [Docker] ❯ git diff js/server.js
diff --git a/js/server.js b/js/server.js
index 7a298cd..09fc338 100644
--- a/js/server.js
+++ b/js/server.js
@@ -69,7 +69,7 @@ function Server(config, callback) {
                        res.status(403).send("This device is not allowed to access your mirror. <br> Please check your config.js or config.js.sample to change this.");
                });
        });
-       app.use(helmet({ contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false }));
+       app.use(helmet({ contentSecurityPolicy: false, crossOriginOpenerPolicy: false, crossOriginEmbedderPolicy: false, crossOriginResourcePolicy: false, originAgentCluster: false, frameg
uard: false }));

        app.use("/js", express.static(__dirname));
```

sdetweil, 2022-05-11:
@CodingCody-1 if u change your js/server.js as suggested, does that provide what u need?

CodingCody-1, 2022-05-13:
Yes, I just made the change and it's working now. Thanks for the help!

sdetweil, 2022-05-13:
@khassel  I guess we need another config parm...

sdetweil, 2022-09-13:
we can close this

closed 2022-09-13 by CodingCody-1