MagicMirrorOrg/MagicMirror #2852: I can't get any further and have no idea what it is! The installation process always stops when installing MagicMirror if «npm» is to be installed. An update from V. 8.5.5 to 8.10.0. All attempts to get this somehow failed, so I would like to ask for an explanation of how I can do the update.

stogli, 2022-05-22:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 14 or later (recommended is 16).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2022-05-22:
how were you installing?  

manual , npm install, or my scooted install?

after you flashed Theo's iange, what else did you do before running mm install?

stogli, 2022-05-24:
Zuerst habe ich das „Automatic Installation Scripts»

 
Verwendedet, danach „ Manual Installation“ und zu guter Letzt, Versucht das npm, vorab zu installieren!
Ich bin ein absoluter Noub und lese mich einfach durch und Versuche.

sdetweil, 2022-05-24:
reflash 
setup ssh
and then use the install script from here

https://github.com/sdetweil/MagicMirror_scripts

does everything for MagicMirror

stogli, 2022-05-26:
I got it to work on an 8GB micro SD, 
but not on the 4GB micro SD! 
Could it be the micro SD? 
I tried different ones and none worked.

sdetweil, 2022-05-26:
no idea. all my SD cards are 16gb or above

df -k

will tell I how much disk is used

stogli, 2022-05-26:
It was probably due to the too small MicroSD 4GB.
After I installed everything on a larger (16 GB) it went without any problems.

sdetweil, 2022-09-13:
we can close this

closed 2022-09-13 by MichMich