MagicMirrorOrg/MagicMirror #2854: Clock Modul MAgicmirror2

stogli, 2022-05-26:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 14 or later (recommended is 16).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

stogli, 2022-05-26:
I want to customize Magicmirror's clock module!
day names
date
time
Who can help me?

sdetweil, 2022-05-27:
generally I bring those things to the forums.

forum.MagicMirror.builders

customize day names? u mean translate?

other things can be done w style sheets, css

stogli, 2022-05-27:
No, no don't translate!
Adjust appearance only.

Friday
05/27/2022
23:56:45

I think the «Stylesheets, CSS» approach will be the way to go, but at the moment I have no idea how to go about it. Or maybe I've already taken the right path!
I would be grateful for a nudge!
I have tried to adjust the «clock.js» file to change the format, unfortunately without success so far.

sdetweil, 2022-05-27:
don't change the js file

see the css file

styles are crazy  you can control everything, but it takes thought about what you want to do

font, font size,  color, and a hundred other things

khassel, 2022-10-02:
no update since May, should be closed @stogli @MichMich

closed 2022-10-03 by MichMich