MagicMirrorOrg/MagicMirror #2856: MMM-NowPlayingOnSpotify

Henry183, 2022-05-28:
The modul is not on the MagicMirror, what should I do ? 

  module: "MMM-NowPlayingOnSpotify",
                           position: "bottom_right",

                           config: {
                           showCoverArt: "true",
                           clientID: "xxxxxxxxxxxxxxxxxxxxxxxxxxx",
                           clientSecret: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
                           accessToken: "BQAJjnPpz6ZUfphJMvCqG9U5NIsvTsq7Y_FwT0scihoiVPUXa5z--DOPCXj28cl2kqsTHLHij4dl2evq1kLHSgmBAXYHUQhCVagekdCwWuAjSNeUzWv$
                           refreshToken: "AQCJ6-mhJlHAJzzmtRnA-LCV5lTW8UMgkJpsVg6fp5Sc7LxcjMvdgJ-q0A_vavDLpk4VcxHctVOokmlu2G2LxAGun7KrDly9TekqEKzn-U248v5uez$

sdetweil, 2022-05-28:
I would move to the MMM-Spotify module... that one is very old
https://github.com/skuethe/MMM-Spotify

rejas, 2022-10-02:
@Henry183 can this be closed now? Or @MichMich just does it since it seems to be not mm2-core related...

khassel, 2022-10-02:
should be closed @MichMich

closed 2022-10-03 by MichMich