MagicMirrorOrg/MagicMirror #2859: When i put a module it wont work

JosefVega, 2022-05-30:
i have git cloned it 
this is an example error 
] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'request'
Require stack:
- /home/pi/MagicMirror/modules/MMM-CoinMarketCap/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js

code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-CoinMarketCap/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',

MichMich, 2022-05-30:
Try `npm install request` from within your MagicMirror folder.

JosefVega, 2022-05-30:
> Try `npm install request` from within your MagicMirror folder.

Thanks it works now :)

sdetweil, 2022-09-13:
@JosefVega can u close this issue?

khassel, 2022-10-02:
should be closed @MichMich

closed 2022-10-03 by MichMich