MagicMirrorOrg/MagicMirror #2860: "1 high severity vulnerability" related to npm during installation

paulsp, 2022-05-31:
Installing MagicMirror per https://docs.magicmirror.builders/getting-started/installation.html#manual-installation , I received the security warning below regarding the npm.

What changes are need in the installation procedure?
```
> magicmirror@2.19.0 prepare
> [ -f node_modules/.bin/husky ] && husky install || echo no husky installed.

no husky installed.

added 265 packages, and audited 266 packages in 1m

20 packages are looking for funding
  run `npm fund` for details

1 high severity vulnerability

To address all issues, run:
  npm audit fix

Run `npm audit` for details.
npm notice 
npm notice New minor version of npm available! 8.5.5 -> 8.11.0
npm notice Changelog: https://github.com/npm/cli/releases/tag/v8.11.0
npm notice Run npm install -g npm@8.11.0 to update!
npm notice 
pi@raspberrypi:~/MagicMirror $ 
```

sdetweil, 2022-05-31:
please ignore the messages. nothing we can do about them. the audit fix causes more trouble than it fixes.

we haven't tested w npm 8.  

next mm release  July 1

paulsp, 2022-05-31:
I request an update to the documentation to ignore the message, as most will apply the security fix.

paulsp, 2022-06-03:
See note on pull request ask why site does not reflect the merge?
https://github.com/MichMich/MagicMirror-Documentation/pull/116#issuecomment-1146006095

rejas, 2022-06-05:
I dont know, maybe @MichMich has to trigger a deployment?

MichMich, 2022-06-05:
The automated build process seems to give an error. Not sure what causes this. Need to take a look at it after this weekend:
```
Wed Jun 1 12:45:01 CEST 2022

[Laravel Forge] Your server is running an older version of Ubuntu (16.04).
We recommend that you provision a new server and manually migrate your sites and resources.
You should not attempt to upgrade a server as this may have unintended side effects.

From github.com:MichMich/MagicMirror-Documentation
 * branch            master     -> FETCH_HEAD
Already up-to-date.
yarn install v1.21.1
warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.2.13: The platform "linux" is incompatible with this module.
info "fsevents@1.2.13" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-darwin-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-darwin-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@2.3.2: The platform "linux" is incompatible with this module.
info "fsevents@2.3.2" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-darwin-arm64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-darwin-arm64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-darwin-arm64@0.14.7: The CPU architecture "x64" is incompatible with this module.
info esbuild-freebsd-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-freebsd-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-android-arm64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-android-arm64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-android-arm64@0.14.7: The CPU architecture "x64" is incompatible with this module.
info esbuild-freebsd-arm64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-freebsd-arm64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-freebsd-arm64@0.14.7: The CPU architecture "x64" is incompatible with this module.
info esbuild-linux-32@0.14.7: The CPU architecture "x64" is incompatible with this module.
info "esbuild-linux-32@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-linux-arm@0.14.7: The CPU architecture "x64" is incompatible with this module.
info "esbuild-linux-arm@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-linux-arm64@0.14.7: The CPU architecture "x64" is incompatible with this module.
info "esbuild-linux-arm64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-linux-mips64le@0.14.7: The CPU architecture "x64" is incompatible with this module.
info "esbuild-linux-mips64le@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-openbsd-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-openbsd-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-netbsd-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-netbsd-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-linux-ppc64le@0.14.7: The CPU architecture "x64" is incompatible with this module.
info "esbuild-linux-ppc64le@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-sunos-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-sunos-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-windows-64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-windows-64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-windows-arm64@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-windows-arm64@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-windows-arm64@0.14.7: The CPU architecture "x64" is incompatible with this module.
info esbuild-windows-32@0.14.7: The platform "linux" is incompatible with this module.
info "esbuild-windows-32@0.14.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info esbuild-windows-32@0.14.7: The CPU architecture "x64" is incompatible with this module.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
success Saved lockfile.
Done in 4.93s.
wait Extracting site metadata...
tip Apply local theme at /home/forge/docs.magicmirror.builders/.vuepress/theme...
tip Apply theme local (extends @vuepress/theme-default) ...
tip Apply plugin container (i.e. "vuepress-plugin-container") ...
[... cut ...]
    at /usr/local/lib/node_modules/vuepress/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:91:9
    at /usr/local/lib/node_modules/vuepress/node_modules/graceful-fs/graceful-fs.js:115:16
    at FSReqCallback.readFileAfterClose [as oncomplete] (node:internal/fs/read_file_context:68:3) {
  opensslErrorStack: [ 'error:03000086:digital envelope routines::initialization error' ],
  library: 'digital envelope routines',
  reason: 'unsupported',
  code: 'ERR_OSSL_EVP_UNSUPPORTED'
}

Node.js v17.7.2
HEAD is now at 7e775cf Merge pull request #116 from paulsp/patch-1
```

khassel, 2022-06-05:
maybe [this](https://dev.to/osmanforhad/react-js-error-message-error-0308010c-digital-envelope-routines-unsupported-23le) this is helpful ...

sdetweil, 2022-06-05:
[Laravel Forge] Your server is running an older version of Ubuntu (16.04).

need to migrate the based test platform

sdetweil, 2022-09-13:
2.21.0 adds new parms to install to turn off audit messages

khassel, 2022-10-02:
there is new documentation with new release v2.21.0, should be closed @paulsp @MichMich

closed 2022-10-03 by MichMich