MagicMirrorOrg/MagicMirror #2863: using config. to initialize a module config varible value fails on node 16.9.1

sdetweil, 2022-06-06:
module config 
```
		module: 'MMM-PrayerTime',
		position: 'top_left',
		config: {
			apiVersion: '1.0', // please, leave unchanged. reserved for future use.
			lat: '29.955271', // latitude of your position (city)
			lon: '31.215639', // longitude of your position (city)
			timezone: "Africa/Cairo", // please refer to http://php.net/manual/en/timezones.php
			timeFormat: 12,
			method: 5,
			playAdzan: ['fajr', 'dhuhr', 'asr', 'maghrib', 'isha'],
			notDisplayed: ['midnight', 'sunset'],
			useUpdateInterval: true,
			updateInterval: 86400 * 1000, // How often do you want to fetch new praying time? (milliseconds)
			animationSpeed: 2.5 * 1000, // Speed of the update animation. (milliseconds)
			language: config.language,  //  ------ this line causes failure 
			showAdzanAlert: true,
			showTomorrow: true,
			vertical: true, // set false for horizontal view
			alertTimer: 15000
			}
		},
```

used to work, 
fails with this error now
```
[05.06.2022 20:58.45.082] [LOG]   Loading config ...
[05.06.2022 20:58.45.088] [ERROR] [31mWARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: ReferenceError: Cannot access 'config' before initialization[39m  <<<<<<<<<< points to language line
[31m    at Object.<anonymous> (/home/sam/MagicMirror.2.17.1/config/config.js:71:14)[39m
[31m    at Module._compile (node:internal/modules/cjs/loader:1116:14)[39m
[31m    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)[39m
[31m    at Module.load (node:internal/modules/cjs/loader:988:32)[39m
[31m    at Module._load (node:internal/modules/cjs/loader:829:12)[39m
[31m    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)[39m
[31m    at Module.require (node:internal/modules/cjs/loader:1012:19)[39m
[31m    at require (node:internal/modules/cjs/helpers:94:18)[39m
[31m    at loadConfig (/home/sam/MagicMirror.2.17.1/js/app.js:69:14)[39m
[31m    at App.start (/home/sam/MagicMirror.2.17.1/js/app.js:212:3)[39m
```

recreate add this module def into config.js
start MM with npm start  

were we depending on some lazy evaluation which doesn't exist anymore?

sdetweil, 2022-06-06:
this fails on 2.18 and 2.19 (the folder is 2.17.1 but upgraded to 2.18)

rejas, 2022-06-06:
what part of that config is the culprit? what does the config check script say?

sdetweil, 2022-06-06:
I don't understand 1st part, I marked the line

        in the module config
             language:  config.language,

config:check finds no issue

rejas, 2022-06-06:
sorry, didnt see that, my bad. do you know when it was the last time this worked?

config:check of course finds no issue since this is valid js, it cant do runtime checking of this (which might be a nice feature if it was possible. is it I ask myself?)

sdetweil, 2022-06-06:
no idea.   I need to look at age of module readme

why doesn't check use require()?

sdetweil, 2022-06-06:
reedme was last updated 2 years ago. so 2.19-8

2.10-2.11 for sure. but this is the first I've seen of this error.  I know the module has been used since then, and there are others.

rejas, 2022-06-08:
> why doesn't check use require()?

do you mean for reading the config js file?

sdetweil, 2022-06-08:
what is different about config:check usage of config.js vs runtime usage?

one works, the other doesn't

khassel, 2023-02-18:
> were we depending on some lazy evaluation which doesn't exist anymore?

I digged a bit into this and found that it has something to do with the changes from `var` to `let`.

If I change the content of `config.js` back to `var config = {` (from `let config = {`) the error is gone. It is still not usable because the `config.language` reference in the module is `undefined` which is may caused by other `var/let` changes in the source code.

closed 2024-10-13 by github-actions[bot]