MagicMirrorOrg/MagicMirror #2867: Question about date module

1Cedric1, 2022-06-11:
Hello and thank you for your job and the possibility to use the MagicMirror!

I have a question regarding the "date" module.

[image]

I would like to have the "day" above the "date", on two lines, instead of one.

Example:

**Monday
4th June 2022**

instead of 

**Monday, 4th June 2022**.

Could you explain me how to do that?

Thank you in advance and have a great week-end.

Kind regards,

Cédric

sdetweil, 2022-06-11:
this is really a forum usage question, not a software issue , see https://forum.magicmirror.builders/

this can be done with a little formatting change and some css..

this is about the css  to do the line split
https://stackoverflow.com/questions/51216748/react-momentjs-date-formatting-with-line-break

and u can specify the dateFormat as one of the clock module config options 
(see the format in the stack overflow topic above, second reply.)
see https://docs.magicmirror.builders/modules/clock.html#configuration-options

but the little 'trick' is the css (using the  white-space: pre;) 

so, in ~/MagicMirror/css/custom.css 
you would add

.clock .date {
   white-space: pre; 
}

1Cedric1, 2022-06-11:
Dear Sam, thank you very much!

best regards

closed 2022-06-11 by 1Cedric1