MagicMirrorOrg/MagicMirror #2869: How do i fix Electron issue. Magic Mirror won't run

kizzytodd, 2022-06-24:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 14 or later (recommended is 16).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2022-06-24:
we are on a later version

npm install electron@17.2.0

what version of node do you have?  must be 14 or 16

kizzytodd, 2022-06-24:
node 18.4.0
npm 8.12.2
node js 16.15.1

kizzytodd, 2022-06-24:
I ran the npm install electron@17.2.0 and got

i@raspberrypi:~/MagicMirror $ npm install electron@17.2.0
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: '@es-joy/jsdoccomment@0.22.2',
npm WARN EBADENGINE   required: { node: '^12 || ^14 || ^16 || ^17' },
npm WARN EBADENGINE   current: { node: 'v18.4.0', npm: '8.12.2' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'eslint-plugin-jsdoc@38.1.6',
npm WARN EBADENGINE   required: { node: '^12 || ^14 || ^16 || ^17' },
npm WARN EBADENGINE   current: { node: 'v18.4.0', npm: '8.12.2' }
npm WARN EBADENGINE }

added 43 packages, removed 58 packages, changed 6 packages, and audited 750 packages in 50s

79 packages are looking for funding
  run `npm fund` for details

3 moderate severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.

kizzytodd, 2022-06-24:
pi@raspberrypi:~/MagicMirror $ npm audit
# npm audit report

got  <11.8.5
Severity: moderate
Got allows a redirect to a UNIX socket - https://github.com/advisories/GHSA-pfrx-2q88-qq97
fix available via `npm audit fix --force`
Will install electron@6.1.12, which is a breaking change
node_modules/got
  @electron/get  *
  Depends on vulnerable versions of got
  node_modules/@electron/get
    electron  >=7.0.0-beta.1
    Depends on vulnerable versions of @electron/get
    node_modules/electron

3 moderate severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force
pi@raspberrypi:~/MagicMirror $

sdetweil, 2022-06-24:
no idea. I suggest u start over, fresh flash, pi os, 32bit, full, called bullseye (current release)

then use my install script, see
https://github.com/sdetweil/MagicMirror_scripts

it does all the setup and dependency install

kizzytodd, 2022-06-24:
So your script will install everything needed (outside the modules and 3rd party modules) to get it going.

closed 2022-06-24 by kizzytodd

sdetweil, 2022-06-24:
correct. get the base up and running. same as manual install would have done... (assuming everything was prepared correctly)