MagicMirrorOrg/MagicMirror #2876: update the magicmirror package in the npm registry

davide125, 2022-07-02:
npm has a magicmirror package, but it's a very old version (2.0.0): https://www.npmjs.com/package/magicmirror . In spite of this, it seems to be actively used and downloaded (22 times in the last 7 days according to the stats). Please consider updating the npm package to the latest tagged version (and ideally keeping it up to date in the future). Thanks!

rejas, 2022-07-03:
Is that a usecase at all: installing MM via npm? or what good would a package at the registry be?

davide125, 2022-07-03:
It makes packaging MM for Linux distros a bit easier (as they often have tooling for importing content from npm). I suspect some users might also end up installing MM from npm out of habit, as it's pretty common for node projects to use it.

rejas, 2023-02-03:
@MichMich could you give some feedback to this issue?

MichMich, 2023-02-03:
That's a good point. I want to limit the amount of ways to install MagicMirror, due to maintenance for the contributors. But if we can find a clean solution to install using npm or yarn, I'm all for it. 

The issue we had back then. Is that you couldn't install it to a specific folder using npm. Not sure if that has changed.

rejas, 2023-02-04:
I can run a `npm i --prefix mm magicmirror` without any problems. Not sure if that was the troublesome usecase but maybe @davide125 can weigh in on how he wants to install it (and maybe check if the current npm can isntall this old npm module)

davide125, 2023-02-05:
In my case I wanted to use npm as the source for packaging magicmirror in Fedora. The packaging tooling in Fedora is designed around pulling sources from npm. It's not mandatory (and I'm working around it at the moment), but it makes things much easier. If you're curious, https://bugzilla.redhat.com/show_bug.cgi?id=2103400 is where the packaging work is happening (that's at an old version for now because it still needs to pass review, I will update it to the latest before importing once it's approved and keep it up to date afterwards).

davide125, 2023-02-12:
Yes, I just checked and it works fine for this. Thanks!

rejas, 2023-02-12:
so might be worth updating the mm package in npm if there's no extra work needed ATM. No guaranteee for the future though, right @MichMich ?

rejas, 2024-10-06:
I still would like to update the npm registry with the newest version, maybe @MichMich can give me the rights there...

jimbo95-beep, 2024-10-06:
Yes

rejas, 2024-10-23:
Published latest version to npm, added step to release docs

KristjanESPERANTO, 2024-10-23:
@rejas Should we add this method to the installation instructions? (Clearly marked as not the default way to do it.)

sdetweil, 2024-10-24:
well we have node dependencies, how do we do that

rejas, 2024-10-31:
> @rejas Should we add this method to the installation instructions? (Clearly marked as not the default way to do it.)

https://github.com/MagicMirrorOrg/MagicMirror-Documentation/pull/270