MagicMirrorOrg/MagicMirror #2879: Unable to update magic mirror

andrew16x, 2022-07-04:
Hello,

I would really appreciate if someone could help me update my magic mirror. I am currently on v 2.13.0 and the little notification that pops up when mm2 is running tells me that there's an update available and that the current installation is 1277 commits behind the master branch. 

I initially tried git pull && npm install which has worked without fail up until recently. 

I also tried 
1 npm update
2 npm install

then
git pull && npm install --only=prod --omit=dev

Also tried  git reset --hard; git clean -df

So far I have been unsuccessful in updating it. 

There aren't any errors during the update process, but it seems that it jumps out prematurely - see output below:

pi@raspberrypi:~/MagicMirror $ git pull && npm install
Updating 3dbe8bfbb..30d5bfe59
pi@raspberrypi:~/MagicMirror $ 

Has anyone experienced a similar issue and can advise how they fixed it? 

Many thanks, 

Andrew

sdetweil, 2022-07-04:
do
```
cd ~/MagicMirror
git status
grep version package.json
node -v
npm -v

u might want to try my upgrade script, see
https://github.com/sdetweil/MagicMirror_scripts

note that upgrade is a two part process

test
then apply

if you are already at 2.20, try with force instead of apply for step 2.

andrew16x, 2022-09-13:
Hi Sam, 

Unfortunately no, I still can't update MM2.
Any other suggestions would be kindly appreciated.

Best regards,
Andrew 

 
 

@andrew16x was this issue resolved?

sdetweil, 2022-09-13:
can you post the ~/MagicMirror/installers/upgrade.log

andrew16x, 2022-09-14:
Hi Sam, 

Please see the log attached as requested. 

Best regards,
Andrew

andrew16x, 2022-09-14:
Not sure why the log file was not attached correctly.  I've pasted its contents below, hopefully that helps:

Upgrade started - Tue Jul  5 16:00:41 BST 2022
system is Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
the os is Distributor ID: Raspbian Description: Raspbian GNU/Linux 9.13 (stretch) Release: 9.13 Codename: stretch
doing test run = true, NO updates will be applied!
[96mCheck current Node installation ...[0m
[0mNode currently installed. Checking version number.
[0mMinimum Node version: [1mv16.9.1[0m
[0mInstalled Node version: [1mv10.6.0[0m
[96mNode should be upgraded.[0m
[91mA Node process is currently running. Can't upgrade.
Please quit all Node processes and restart the update.
pi 9445 4013 0 Jul04 pts/0 00:00:02 node /usr/bin/pm2 logs mm pi 10416 4013 0 Jul04 pts/0 00:00:02 node /usr/bin/pm2 logs mm

Upgrade started - Tue Jul  5 16:03:43 BST 2022
system is Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
the os is Distributor ID: Raspbian Description: Raspbian GNU/Linux 9.13 (stretch) Release: 9.13 Codename: stretch
doing test run = true, NO updates will be applied!
[96mCheck current Node installation ...[0m
[0mNode currently installed. Checking version number.
[0mMinimum Node version: [1mv16.9.1[0m
[0mInstalled Node version: [1mv10.6.0[0m
[96mNode should be upgraded.[0m
[92mNode.js upgrade defered, doing test run[0m
[96mCheck current NPM installation ...[0m
[0mNPM currently installed. Checking version number.
[0mMinimum npm version: [1mV7.11.2[0m
[0mInstalled npm version: [1mV6.1.0[0m
[96mnpm should be upgraded.[0m
[96mnpm upgrade defered, doing test run  ...[90m

saving custom.css
remote name = origin
upgrading from version 2.13.0 to 2.20.0
fetching latest revisions
git fetch rc=0
current branch = master
On branch master
Your branch is behind 'origin/master' by 1277 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   package-lock.json
	modified:   package.json
	modified:   vendor/package-lock.json

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	css/save_custom.css

no changes added to commit (use "git add" and/or "git commit -a")
there are 3 local files that are different than the master repo

   package-lock.json
   package.json
   vendor/package-lock.json

save/restore files selection = y
erasing lock files
test merge result rc='' , if empty, no conflicts
skipping merge, only test run
Checking for modules with removed libraries
downloading dumpactivemodules script

processing dependency changes for active modules with package.json files
processing for module MMM-Remote-Control please wait
----------------------------------
skipped processing for MMM-Remote-Control, doing test run
processing complete for module MMM-Remote-Control
processing for module MMM-Wunderlist-Enhanced please wait
----------------------------------
skipped processing for MMM-Wunderlist-Enhanced, doing test run
processing complete for module MMM-Wunderlist-Enhanced
restoring custom.css
removing git alias
test run, restoring files stashed
Upgrade ended - Tue Jul  5 16:05:01 BST 2022

sdetweil, 2022-09-14:
>doing test run = true

upgrade is a two step process

test run by default. check to see if any issues, don't crater the users system

and another with the word apply added on to DO the upgrade

andrew16x, 2022-09-14:
Just used the script with the "apply" at the end this time, it completed, however when I started MM2, the notification was still present at the top to say that MM2 needs an update and that is behind 'origin/master' by 1277 commits. So I rebooted pi thinking that was just a glitch. Normally after a reboot MM2 would start up automatically but not this time. I tried to start it manually and got this:

`pi@raspberrypi:~/MagicMirror $ pm2 start MagicMirror
[PM2] Spawning PM2 daemon with pm2_home=/home/pi/.pm2
[PM2] PM2 Successfully daemonized
[PM2][ERROR] script not found : /home/pi/MagicMirror/MagicMirror
script not found : /home/pi/MagicMirror/MagicMirror
`
What could have happened?

sdetweil, 2022-09-14:
show me the ~/MagicMirror/installers/upgrade.log

short term
open a terminal window, ctrl-alt-t,or use ssh to connect
```
 cd ~/MagicMirror
npm start
```

andrew16x, 2022-09-14:
`npm start` does work indeed for the time being. 

Wondering if I need to recreate the startup script as per [https://docs.magicmirror.builders/configuration/autostart.html#using-pm2](url). It might have been deleted due to the upgrade and because when I installed MM2 years ago, I actually created it inside the MagicMirror folder. The documentation above suggests the startup scrip should be placed outside of the MM2 folder.

Also intrigued as to why it thinks it still needs an update... I've attached the log file, hopefully you can see it now, if not, I'll copy paste the output.
[upgrade.log](https://github.com/MichMich/MagicMirror/files/9566392/upgrade.log)

sdetweil, 2022-09-14:
you can run the fixuppm2 script from my scripts

https://github.com/sdetweil/MagicMirror_scripts

it will set up pm2 again.. 

but u need to delete the existing one

pm2 status 
pm2 delete number or name
pm2 save -f

then run my script..

why still behind? 
says there is a merge conflict .. 

do git status , show results 

also, you are on pi OS stretch , which cannot be upgraded to latest required nodejs version...

andrew16x, 2022-09-14:
here is the output for  `git status`

`pi@raspberrypi:~/MagicMirror $ git status
On branch master
Your branch is behind 'origin/master' by 1277 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	deleted:    fonts/package-lock.json
	deleted:    package-lock.json
	modified:   package.json
	deleted:    vendor/package-lock.json

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	css/icons/
	installers/stashed_files

no changes added to commit (use "git add" and/or "git commit -a")
`

in terms of the OS version, can I update to a later version without impacting MM2?

sdetweil, 2022-09-14:
>in terms of the OS version, can I update to a later version without impacting MM2?

maybe

but I think safer to save and rebuild
download my https://github.com/sdetweil/MagicMirror-backup-restore

and run mm_backup
it will create a folder ~/MM_backup
with 3 files
config.js
custom.css
and a file with the links to the modules installed

save that folder to another place

build a new OS sd card
use my script to install MM
https://github.com/sdetweil/MagicMirror_scripts

then restore the MM-backup folder to the new disk
and clone my backup_restore repo again

then run MM_restore
it will copy the config.js, custom.css back
and reinstall all the modules you had installed, doing npm install on each as required. 

this is the least worst way..   and u haven't changed your existing sd card 

on the merge update problem, I think this is a git problem(as I had it a couple years ago too) , as you are on a very old platform
do 
git -v 
(mine is 2.37.1)
you may have to update  git to get the update installed..  (google search)

andrew16x, 2022-09-15:
Quick update, I have completed the backup using the mm_backup.sh script and confirm that I can see the three files created in the MM-backup folder. Copied these onto my NAS for the time being. 

I've also ordered a new 32GB SD card (should arrive Sat) on which I am planning to install the latest RP OS, which I believe it's Bullseye. 

I'm fairly clear with the rest of instructions above apart from the step "clone my backup_restore repo again" - what exactly am I meant to be doing there please?

sdetweil, 2022-09-15:
u have to get the restore script onto the new card

andrew16x, 2022-09-22:
Hello, I have successfully managed to install the latest OS on the new card, installed MM and restored the backup, all seems to be up to date now. There were two weather modules which were obsolete that I didn't know of before so had to revert to the default weather module. Seems like the updated OS has sorted my initial problem so I'm happy to close off this issue. Many thanks again @sdetweil for all your help and guidance.

sdetweil, 2022-09-22:
awesome, thanks for the update

sdetweil, 2022-09-22:
I updated the backup/restore to all ow running them from the web, so you don't have to git clone the repo anymore. and restore can fetch your backup repo for you..  thanks for pushing me to update

andrew16x, 2022-09-22:
No worries, thanks again. Do you know by any chance if the Alexa module is now functioning correctly? I know there were lots of incompatibility issues when I tried this couple of years ago.

sdetweil, 2022-09-22:
I do not. 

but recently Amazon implemented new 'branding' certification which disables using skills if not certified. no software only solution can pass the certification process

khassel, 2022-10-02:
seems to be solved, can this be closed? @andrew16x @MichMich

andrew16x, 2022-10-02:
Yes, it can be closed, thank you.

closed 2022-10-03 by MichMich