MagicMirrorOrg/MagicMirror #2880: safer weather template by adding a remote config variable

Kayakbabe, 2022-07-06:
_I posted this same topic on the forum. But putting it here as I think this might be a helpful feature to add to the default weather module.

Changing the code for the module to allow for a remote template file will not break the weather module if there is no config variable or no remote file. If not remote template is setup then the module will just work as before.

Forum link: [https://forum.magicmirror.builders/topic/16957/safer-weather-template-by-adding-a-remote-config-variable?_=1657147849733](https://forum.magicmirror.builders/topic/16957/safer-weather-template-by-adding-a-remote-config-variable?_=1657147849733)
_

The compliments module is great. I love the remoteFile config variable so that I can add my own custom list of special date messages and other things. The beauty is that a custom file of compliments doesn’t get erased upon updating the mm and is easy to copy to another mirror instance. It is also easy to find to add or alter the compliments.

The weather module uses external template files like current.njk and forcast.njk. It is possible to modify them to make the weather appear as we like. But they are at risk of being overwritten by an update to the mirror.

I think that having a weather module config variable for a custom template would make updating less stressful and easier.

To that end I have modified my local weather module instance as follows.

In the weather module weather.js defaults section I added

```
	defaults: {
		remoteTemplate: "current.njk",
```
and I altered the getTemplate function to the following:

```
	// Select the template depending on the display type.
	getTemplate: function () {
		
		if (this.config.remoteTemplate !="") {return this.config.remoteTemplate}
		 else {
		
			switch (this.config.type.toLowerCase()) {
				case "current":
					return "current.njk";
				case "hourly":
					return "hourly.njk";
				case "daily":
				case "forecast":
					return "forecast.njk";
				//Make the invalid values use the "Loading..." from forecast
				default:
					return "forecast.njk";
			}
	   }
	},
```
Thus, if I do not create or set a remote template, the module will not fail, it will get the template it should.
But I want to use a custom template becuase I want control to the css tags used in the html so I can more easily customise the weather. I also wanted a different layout. So made copies of current.njk and forecast.njk and put them in the mm config folder. I also renamed them to be very clear they were my custom template.
Then,
in the weather section of my config.js
I added the relative path to my custom template file.
```
config: {
  remoteTemplate: "../../../../config/kellyforecast.njk",
```
I actually have a custom version of current and forecast templates. I can quickly tweak the templates and not get into the folder structure of the mm default modules.

I would like to contribute this to the mm repository.

sdetweil, 2022-10-05:
we should close one of these as a dup

rejas, 2022-10-05:
agreed. maybe one day @MichMich will give us issue-handling-rights ;-) (not sure if github allows such fine grained access rights though)

khassel, 2022-10-05:
> not sure if github allows such fine grained access rights though
see https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-personal-account-settings/permission-levels-for-a-personal-account-repository

MichMich, 2022-10-06:
I invited you three als collaborators. :) Without you three MagicMirror² wouldn't be what it is now. <3

rejas, 2022-10-06:
Awwww, thanks for the flowers (and the invite). Will try to not break your trust (and your code :-)

closed 2022-10-06 by rejas

khassel, 2022-10-06:
> Will try to not break your trust (and your code :-)

me too ... currently I'm working with 2 git remotes on my machine, have from now on to double check not to push to the wrong repo ...

rejas, 2022-10-06:
I'll probably still be using my own fork for workinprogress. That way the main repo doesnt get trashed by my "late night ideas that never get finished"-branches :-) (plus I dont push in the wrong repo :-D

khassel, 2022-10-06:
meanwhile I configured a dummy-push-url for this repo ... ;)

sdetweil, 2022-10-06:
eh?

khassel, 2022-10-06:
o.k., more detailed ...

you can work with more than one remote repository. E.g. I have 2 configured, my own fork of this repo **and** this repo.

So I can work on the branches of my fork but can also fetch new stuff from the original repo.

So far it was no problem if I tried to push accidentially to this repo because I got `permission denied`. Now we have these permissions ...

Therefore I configured git in this way:

```bash
$ git remote -v                                           
k13     git@github.com:khassel/MagicMirror.git (fetch)    
k13     git@github.com:khassel/MagicMirror.git (push)     
origin  git@github.com:MichMich/MagicMirror.git (fetch)   
origin  fake@github.com (push)                            
```