MagicMirrorOrg/MagicMirror #2882: Execution fails with other user than pi

cristinabolanospeno, 2022-07-19:
Hi! I am trying to run MagicMirror with other user than pi or the superuser created on the first configuration of a Raspbian SD (as for the new releases the default user pi does not exist).

## Steps to reproduce

1. Clone MagicMirror and install it.
    ```
    git clone --branch v2.18.0 https://github.com/MichMich/MagicMirror magicmirror
    cd magicmirror
    npm install --only=prod --omit=dev
    npm audit fix
    cd ..
    ```
3. Create user and group. Btw I have also tried the `--system` option for the `adduser` command.
    ```
    $ addgroup magicmirror
    $ adduser --home $(realpath magicmirror) --no-create-home \
        --disabled-password --shell /bin/bash --ingroup magicmirror \
        --gecos "MagicMirror user" magicmirror
    ```
4. Set ownership of magicmirror to magicmirror user.
    ```
    sudo chown -R magicmirror:magicmirror magicmirror/
    ```
6.  Log into the user with `sudo su - magicmirror` (it has no password set).
7.  Set sample config with `cp config/config.js.sample config/config.js`.
8.  Execute `npm run start`.

With this, I expect a normal execution of the MagicMirror.

## Actual results

Console output:

```shell
magicmirror@raspberry:~$ npm run start

> magicmirror@2.18.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[19.07.2022 11:35.59.810] [LOG]   Starting MagicMirror: v2.18.0
[19.07.2022 11:35.59.818] [LOG]   Loading config ...
[19.07.2022 11:35.59.824] [LOG]   Loading module helpers ...
[19.07.2022 11:35.59.828] [LOG]   No helper found for module: alert.
[19.07.2022 11:35.59.846] [LOG]   Initializing new module helper ...
[19.07.2022 11:35.59.849] [LOG]   Module helper loaded: updatenotification
[19.07.2022 11:35.59.851] [LOG]   No helper found for module: clock.
[19.07.2022 11:36.00.004] [LOG]   Initializing new module helper ...
[19.07.2022 11:36.00.006] [LOG]   Module helper loaded: calendar
[19.07.2022 11:36.00.009] [LOG]   No helper found for module: compliments.
[19.07.2022 11:36.00.012] [LOG]   No helper found for module: weather.
[19.07.2022 11:36.00.038] [LOG]   Initializing new module helper ...
[19.07.2022 11:36.00.040] [LOG]   Module helper loaded: newsfeed
[19.07.2022 11:36.00.042] [LOG]   All module helpers loaded.
[19.07.2022 11:36.00.145] [LOG]   Starting server on port 8080 ... 
[19.07.2022 11:36.00.161] [LOG]   Server started ...
[19.07.2022 11:36.00.162] [LOG]   Connecting socket for: updatenotification
[19.07.2022 11:36.00.164] [LOG]   Starting module helper: updatenotification
[19.07.2022 11:36.00.164] [LOG]   Connecting socket for: calendar
[19.07.2022 11:36.00.166] [LOG]   Starting node helper for: calendar
[19.07.2022 11:36.00.167] [LOG]   Connecting socket for: newsfeed
[19.07.2022 11:36.00.168] [LOG]   Starting node helper for: newsfeed
[19.07.2022 11:36.00.169] [LOG]   Sockets connected & modules started ...
No protocol specified
/home/pi/magicmirror/node_modules/electron/dist/electron exited with signal SIGTRAP
npm ERR! code 1
npm ERR! path /home/pi/magicmirror
npm ERR! command failed
npm ERR! command sh -c DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/magicmirror/.npm/_logs/2022-07-19T09_36_00_248Z-debug.log
magicmirror@raspberry:~$ 
```

And the log:

```shell
0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'start' ]
1 info using npm@7.5.2
2 info using node@v12.22.12
3 timing config:load:defaults Completed in 6ms
4 timing config:load:file:/usr/share/nodejs/npm/npmrc Completed in 15ms
5 timing config:load:builtin Completed in 16ms
6 timing config:load:cli Completed in 6ms
7 timing config:load:env Completed in 1ms
8 timing config:load:project Completed in 2ms
9 timing config:load:file:/home/pi/magicmirror/.npmrc Completed in 1ms
10 timing config:load:user Completed in 1ms
11 timing config:load:file:/etc/npmrc Completed in 0ms
12 timing config:load:global Completed in 1ms
13 timing config:load:cafile Completed in 1ms
14 timing config:load:validate Completed in 1ms
15 timing config:load:setUserAgent Completed in 3ms
16 timing config:load:setEnvs Completed in 4ms
17 timing config:load Completed in 43ms
18 verbose npm-session 5308ddf2bb99a241
19 timing npm:load Completed in 82ms
20 timing command:run-script Completed in 1615ms
21 verbose stack Error: command failed
21 verbose stack     at ChildProcess.<anonymous> (/usr/share/nodejs/@npmcli/promise-spawn/index.js:64:27)
21 verbose stack     at ChildProcess.emit (events.js:314:20)
21 verbose stack     at maybeClose (internal/child_process.js:1022:16)
21 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
22 verbose pkgid magicmirror@2.18.0
23 verbose cwd /home/pi/magicmirror
24 verbose Linux 5.15.32-v7l+
25 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
26 verbose node v12.22.12
27 verbose npm  v7.5.2
28 error code 1
29 error path /home/pi/magicmirror
30 error command failed
31 error command sh -c DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
32 verbose exit 1
```

## Additional information

The error ChildProcess does not help me much tough. I thought it could be a 'privilege' problem, so I tried giving sudo privileges to magicmirror user and disabling the password, just as the pi user, with no luck. Maybe the problem is related to the line `No protocol specified`?

Also, I know I am using older versions of `npm` and `MagicMirror`, because I am using the repository version of npm (12.22.5 as you can see in the log). But the thing is that this works. If I execute `npm run start` with the pi user, without changing the ownership of the files, it works perfectly!

P.S. I need that user for my project, so "just run it as pi" will not work for me :(

sdetweil, 2022-07-19:
well, first, I would wait til I created the user before doing the git clone (which needs to be under their permission level)

never run audit fix... causes more problems than it solves.. 

the real issue is 

23 verbose cwd /home/pi/magicmirror

somewhere there is a hard coded /home/pi

but this is 2 releases back (2.18) ... 2.20 is current

will look some more.

sdetweil, 2022-07-19:
I have tested with the latest raspios full, creating the user first (on 1st boot), all works as expected

cristinabolanospeno, 2022-07-20:
But you are creating the user when configuring Raspbian for the first time right? Like you say, first boot. I need to create a second user, which will be the one running MagicMirror. I don't know the exact difference in privileges between the two users, I mentioned giving sudo privileges does not work...

I am running this on a Raspberry Pi 4 following the tutorials, with graphical desktop. A normal installation I guess, except the new user part. I will try with the new version 2.20, but I suspect this is not a upgrade problem, but a privileges one. I will keep you updated! And thanks for the fast answer :)

cristinabolanospeno, 2022-07-20:
It does not run on 2.20, but it outputed a different error. I will leave the steps to reproduce and the output. Also, it still works with the pi user (magicmirror ownership).

```shell
pi@magicmirror:~$ sudo deluser --remove-home magicmirror
pi@magicmirror:~$ git clone --branch v2.20.0 https://github.com/MichMich/MagicMirror magicmirror
pi@magicmirror:~$ sudo addgroup magicmirror
pi@magicmirror:~$ sudo adduser --home $(realpath magicmirror) --no-create-home --disabled-password --shell /bin/bash --ingroup magicmirror --gecos "MagicMirror user" magicmirror
pi@magicmirror:~$ sudo chown -R magicmirror:magicmirror magicmirror/
pi@magicmirror:~$ sudo su - magicmirror

magicmirror@magicmirror:~$ npm install --only=prod --omit=dev
magicmirror@magicmirror:~$ npm run start
```

```shell
> magicmirror@2.20.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[20.07.2022 12:37.51.969] [LOG]   Starting MagicMirror: v2.20.0
[20.07.2022 12:37.51.977] [LOG]   Loading config ...
[20.07.2022 12:37.51.983] [LOG]   Loading module helpers ...
[20.07.2022 12:37.51.986] [LOG]   No helper found for module: alert.
[20.07.2022 12:37.52.002] [LOG]   Initializing new module helper ...
[20.07.2022 12:37.52.003] [LOG]   Module helper loaded: updatenotification
[20.07.2022 12:37.52.005] [LOG]   No helper found for module: clock.
[20.07.2022 12:37.52.202] [LOG]   Initializing new module helper ...
[20.07.2022 12:37.52.204] [LOG]   Module helper loaded: calendar
[20.07.2022 12:37.52.205] [LOG]   No helper found for module: compliments.
[20.07.2022 12:37.52.206] [LOG]   No helper found for module: weather.
[20.07.2022 12:37.52.228] [LOG]   Initializing new module helper ...
[20.07.2022 12:37.52.228] [LOG]   Module helper loaded: newsfeed
[20.07.2022 12:37.52.229] [LOG]   All module helpers loaded.
[20.07.2022 12:37.52.343] [LOG]   Starting server on port 8080 ... 
[20.07.2022 12:37.52.358] [LOG]   Server started ...
[20.07.2022 12:37.52.359] [LOG]   Connecting socket for: updatenotification
[20.07.2022 12:37.52.360] [LOG]   Starting module helper: updatenotification
[20.07.2022 12:37.52.360] [LOG]   Connecting socket for: calendar
[20.07.2022 12:37.52.361] [LOG]   Starting node helper for: calendar
[20.07.2022 12:37.52.362] [LOG]   Connecting socket for: newsfeed
[20.07.2022 12:37.52.362] [LOG]   Starting node helper for: newsfeed
[20.07.2022 12:37.52.363] [LOG]   Sockets connected & modules started ...
No protocol specified
[21242:0720/123752.368153:ERROR:ozone_platform_x11.cc(248)] Missing X server or $DISPLAY
[21242:0720/123752.368268:ERROR:env.cc(226)] The platform failed to initialize.  Exiting.
/home/pi/magicmirror/node_modules/electron/dist/electron exited with signal SIGSEGV
```

```
0 verbose cli /usr/bin/node /usr/bin/npm
1 info using npm@8.11.0
2 info using node@v16.16.0
3 timing npm:load:whichnode Completed in 0ms
4 timing config:load:defaults Completed in 4ms
5 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 2ms
6 timing config:load:builtin Completed in 2ms
7 timing config:load:cli Completed in 4ms
8 timing config:load:env Completed in 0ms
9 timing config:load:project Completed in 9ms
10 timing config:load:file:/home/pi/magicmirror/.npmrc Completed in 0ms
11 timing config:load:user Completed in 0ms
12 timing config:load:file:/usr/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 1ms
14 timing config:load:validate Completed in 0ms
15 timing config:load:credentials Completed in 3ms
16 timing config:load:setEnvs Completed in 2ms
17 timing config:load Completed in 29ms
18 timing npm:load:configload Completed in 31ms
19 timing npm:load:mkdirpcache Completed in 2ms
20 timing npm:load:mkdirplogs Completed in 1ms
21 verbose title npm run start
22 verbose argv "run" "start"
23 timing npm:load:setTitle Completed in 3ms
24 timing config:load:flatten Completed in 8ms
25 timing npm:load:display Completed in 14ms
26 verbose logfile logs-max:10 dir:/home/pi/magicmirror/.npm/_logs
27 verbose logfile /home/pi/magicmirror/.npm/_logs/2022-07-20T10_37_50_373Z-debug-0.log
28 timing npm:load:logFile Completed in 13ms
29 timing npm:load:timers Completed in 1ms
30 timing npm:load:configScope Completed in 0ms
31 timing npm:load Completed in 66ms
32 silly logfile done cleaning log files
33 timing command:run Completed in 1976ms
34 verbose exit 1
35 timing npm Completed in 2063ms
36 verbose code 1
```

sdetweil, 2022-07-20:
sorry  still one question..  what is wrong with the userid created at raspi boot time?

closed 2022-07-20 by cristinabolanospeno