MagicMirrorOrg/MagicMirror #2883: The default NewsFeed module does not respect the rss TTL value

bootstrap-prime, 2022-07-19:
## I found a bug in MagicMirror

Browser: Firefox
Hardware: Linux, NixOS
Node version: v16.15.0
MagicMirror: v2.20.0

Description:
The default NewsFeed module does not respect the rss TTL (time-to-live) value. Providers can set this to request that feed readers check every N minutes, which can ease strain on their services. (See [xeaiso.net](https://xeiaso.net/blog.rss) for an example of a provider that sets this ttl value).
It would be helpful if MagicMirror supported this.

sdetweil, 2022-07-19:
you mean too frequent? ('too high', but actually too low, number)

bootstrap-prime, 2022-07-20:
sorry, yes

khassel, 2022-07-20:
what do you mean with "support this".

The ttl value is accessible in the module, so we could do something, but what? Should we increase the updateInterval if it is below ttl?

bootstrap-prime, 2022-07-21:
Yeah, essentially.
I think the best approach would probably be to make reloadInterval an Option(ms) type, (or the js equivalent).
if you set reloadInterval, it is whatever you set it to. If reloadInterval is unset, it is the default (which would be bumped up to every 30mins) or the ttl (if present).

sdetweil, 2022-07-21:
reloadInterval is already a configurable option, default 5 mins.

anyone could change it in their config.

it could be overridden by the feed ttl

khassel, 2022-07-21:
the above PR uses now ttl for reloadInterval if submitted by the feed (and if < 24h as mentioned in the docs).

The default reloadInterval is not changed, because I think this is not necassary. You can set a reloadInterval globally for the newsfeed module and for each feed (which should be sufficient). Changing the default value would affect all installations which rely on the current defaults.

sdetweil, 2022-07-21:
thank you for logging that  Feed TTL overrides the config option

khassel, 2022-10-01:
fixed in new release v2.21.0, should be closed @bootstrap-prime @MichMich

PR #2884 (2022-07-28): respect rss ttl provided by newsfeed
Fixes #2883
closed 2022-10-02 by MichMich