MagicMirrorOrg/MagicMirror #2885: Calendar Repeating Events one day off

DAHines57, 2022-07-23:
Hardware: RPi 3, Stretch
Node version: v16.14.2
MagicMirror: v2.20.0

Description: 
I have repeating google calendar event set for Thursdays at 8pm to 11pm, and this displays correctly there, but in the calendar module this event is shown to be on Friday from 8 to 11pm.  I have also noticed if I set the event to be 6pm to 8pm, the event displays correctly in the module.

Timezone: America/Chicago

iCal:
```
BEGIN:VEVENT
DTSTART;TZID=America/Chicago:20220728T200000
DTEND;TZID=America/Chicago:20220728T230000
RRULE:FREQ=DAILY;COUNT=30;INTERVAL=7
DTSTAMP:20220723T162624Z
```

Calendar Config:
```{
			module: "calendar",
			header: "Calendar",
			position: "bottom_left",
			disabled: false,
			config: {
				showEnd: false,
				dateFormat: "MMM DD - hh:mm A",
				maximumEntries: 6,
				calendars: [
```

Screenshot:
[image]

Thanks!

sdetweil, 2022-07-25:
@DAHines57 what did u do to fix this?

DAHines57, 2022-07-25:
@sdetweil I had noticed that my repository was acting strange, so I re-cloned the repo and tried again. However I did notice that when I create the recurring event from my iPhone the date is still off. But if I create the event directly in google calendar the dates show correctly.

sdetweil, 2022-07-25:
ok, thanks. I will look at this.  is the ve ent above a good example?

DAHines57, 2022-07-25:
Here's a more updated example, the first was created directly in gCal ('Test Gcal'), the second using the iphone default calendar app ('Test') .

```
BEGIN:VEVENT
DTSTART;TZID=America/Chicago:20220721T200000
DTEND;TZID=America/Chicago:20220721T230000
RRULE:FREQ=WEEKLY;BYDAY=TH
DTSTAMP:20220725T015743Z
UID:
CREATED:20220725T005851Z

BEGIN:VEVENT
DTSTART;TZID=America/Chicago:20220728T210000
DTEND;TZID=America/Chicago:20220728T220000
RRULE:FREQ=WEEKLY;BYDAY=TH
DTSTAMP:20220725T015743Z
UID:
CREATED:20220725T014034Z
```
[image]

DAHines57, 2022-07-25:
@sdetweil should I reopen this?

sdetweil, 2022-07-25:
this is a bug in the repeating rule processor we use 
( my clock is set to Chicago as well) 

we asked for the right range, 
now  Mon Jul 25 2022 08:04:54 GMT-0500
to (year from now) 
Mon Jul 24 2023 23:59:59 GMT-0500

and it skipped 7/28, returning 08/04 as the 1st date 
 ["2022-08-04T02:00:00.000Z",

```
Search for recurring events between: Mon Jul 25 2022 08:04:54 GMT-0500 (Central Daylight Time) and Mon Jul 24 2023 23:59:59 GMT-0500 (Central Daylight Time)
[25.07.2022 08:04.54.634] [DEBUG] Title: start 28, with dates: ["2022-08-04T02:00:00.000Z","2022-08-11T02:00:00.000Z","2022-08-18T02:00:00.000Z"
```

DAHines57, 2022-07-25:
Interesting, makes sense why just starting the event a week earlier "solves" the problem

khassel, 2022-10-02:
seems to be fixed in v2.21.0, can this be closed @sdetweil ?

sdetweil, 2022-10-02:
just waiting for confirmation

DAHines57, 2022-10-03:
@sdetweil It's working for me after pulling v2.21.0

sdetweil, 2022-10-03:
awesome. can u close this issue now.

PR #2905 (2022-09-06): Fixcal
rrule and luxon miscalculate event day when byday is specified, but NOT without. 

correct adjustment in calendarutils.js to adjust only when byday is specified

fixes: #2885
closed 2022-10-03 by DAHines57