MagicMirrorOrg/MagicMirror #2887: Failed to load swrast driver

paniman, 2022-07-24:
Hi Guys,
I get the following error on startup:

```
[24.07.2022 17:07.02.966] [LOG]   Connecting socket for: newsfeed
[24.07.2022 17:07.02.967] [LOG]   Starting node helper for: newsfeed
[24.07.2022 17:07.02.967] [LOG]   Sockets connected & modules started ...
[24.07.2022 17:07.03.391] [LOG]   Launching application.
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open zink: /usr/lib/dri/zink_dri.so: Kann die Shared-Object-Datei nicht öffnen: Keine Berechtigung (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: zink
MESA-LOADER: failed to open kms_swrast: /usr/lib/dri/kms_swrast_dri.so: Kann die Shared-Object-Datei nicht öffnen: Keine Berechtigung (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: Kann die Shared-Object-Datei nicht öffnen: Keine Berechtigung (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load swrast driver
electron exited with signal SIGINT
```

This happened when I pulled the latest commits (I think my last pull was one month ago).

Can you please help me?
If you need any further information, please let me know.

Best,
Steve

khassel, 2022-07-24:
I read already your post in the forum but cannot reproduce this on my pi4 (64 bit image).

With v2.20 we itnroduced a new electron version (v19, before v17).

How did you update? Only `git pull` is not sufficient, you have to run `npm install --only=prod --omit=dev` too.

paniman, 2022-07-25:
Beforehand I used a Tinkerboard S with an Armbian Distribution.

But now I have the same issue when I use the Magic Mirror OS with an Raspberry PI 2 model B:

[25.07.2022 16:54.58.499] [LOG]   Launching application.
MESA-LOADER: failed to retrieve device information
gbm: failed to open any driver (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/vc4_dri.so: cannot open shared object file: No such file or directory
failed to load driver: vc4
gbm: failed to open any driver (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory
failed to load driver: kms_swrast
gbm: failed to open any driver (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
failed to load swrast driver

sdetweil, 2022-07-25:
what is output of 

uname -a

paniman, 2022-07-25:
> what is output of
> 
> uname -a

`pi@magicmirroros:~/magicmirror/mounts/config $ uname -a
Linux magicmirroros 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
`

sdetweil, 2022-07-25:
looks like you will have to install the missing driver..  at least its armv7l like pi3/4

paniman, 2022-07-26:
Hmm okay, so I just wanted to let you know that I gave up and re-installed the system and hope that this error won't come back :) Thanks for your support guys!

closed 2022-07-26 by paniman