MagicMirrorOrg/MagicMirror #2892: All modules with embedding web pages do not work anymore

MasterLH, 2022-07-26:
This is my first time ever, that I open a issue on github, so I hope that I do everything correct.
With the newest Magic Mirror version 2.20 none of my modules regarding webpage embedding do work anymore...
I tried out MMM-Iframe, MMM-SmartWebDisplay and MMM-yframe, as well as MMM-iFrame-Ping, MMM-WebViewer and so on.

Opening webpages like "google.com" sometimes does work, but for example my local home-assistant lovelace dashboards do not work. Magic mirror crashes then and the screen becomes black. However if I open up Magic Mirror via the browser (localhost:8080) everything works fine somehow...

Can you please help me to fix this?

MasterLH, 2022-07-26:
The log always shows an error message like this:
[87:0726/122213.513985:ERROR:broker_posix.cc(46)] Received unexpected number of handles

When googling I found out, that it may has something to do with allowed memory usage of the browser when using docker...
My Magic Mirror instance is running on docker via https://gitlab.com/khassel/magicmirror

sdetweil, 2022-07-26:
@MasterLH you should open an issue on that GitHub repo
@khassel

MasterLH, 2022-07-27:
Fixed with the help of: https://khassel.gitlab.io/magicmirror/faq/#running-on-a-raspberry-pi-ends-with-a-white-or-black-screen-after-a-while

Ty so much @khassel <3

closed 2022-07-27 by MasterLH