MagicMirrorOrg/MagicMirror #2896: Remove unneeded (and unwanted) '.' after the year in calendar repeatingCountTitle

mcavoya, 2022-08-05:
Platform: Raspberry Pi 3 (browser/Electron version agnostic)
Node Version: 16.16.0
MagicMirror² Version: 2.20.0
Description: Calendar shows a '. ' after the year in repeatingCountTitle
Steps to Reproduce: Load a calendar with repeating events. e.g. Joe's birthday, with a start date seven years ago, that repeats yearly.
Expected Results: "Joe's Birthday, 7 years"
Actual Results: "Joe's Birthday, 7. years"
Configuration: see attached [config.zip](https://github.com/MichMich/MagicMirror/files/9268902/config.zip)
Additional Notes: The fix is simple. 
Change line 265 in calendar.js
```
from: repeatingCountTitle = ", " + yearDiff + ". " + repeatingCountTitle;
to:   repeatingCountTitle = ", " + yearDiff + repeatingCountTitle;
```

In cases where puncutation is desired after yearDiff, the config file can simply add it in the repeatingCountTitle.

rejas, 2022-08-09:
Thx for the report @mcavoya It indeed seems a little off when looking at the result in english. 
Since @MichMich wrote that line (way back in 2016) maybe he had some other i18n format in mind?

khassel, 2023-01-13:
I think this should be fixed if there comes no veto by @MichMich

MichMich, 2023-01-14:
All fine by me. Not emotionally attached to the `.`. ;)

mcavoya, 2023-04-07:
I do not believe this topic should be marked Closed.

The code was refactored, but the offending '.' character remains.  It appears right after {yearDiff}.

`repeatingCountTitle = ', ${yearDiff}. ${repeatingCountTitle}';`

And it can be seen here as "31. Years"

[image]

khassel, 2023-04-07:
you are right, sadly it was reintroduced by another PR:

[image]

So I will reopen this issue and fix it again with a new PR ...

mcavoya, 2023-04-07:
Thanks. I can only imagine that's a frequent squence of events.

khassel, 2023-07-01:
fixed with new release `v2.24.0`

PR #3003 (2023-01-14): removed unneeded (and unwanted) '.' in calendar
- removed unneeded (and unwanted) '.' after the year in calendar repeatingCountTitle (fixes #2896)
- update Collaboration.md
commit d2327d3d (2023-01-14): removed unneeded (and unwanted) '.' in calendar (#3003)

- removed unneeded (and unwanted) '.' after the year in calendar
repeatingCountTitle (fixes #2896)
- update Collaboration.md
closed 2023-07-01 by khassel