MagicMirrorOrg/MagicMirror #2899: weather module, weatherbit provider doesn't use type to set the endpooint, forces user..

sdetweil, 2022-08-15:
all in title 
```
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				updateInterval: 15*1000,
				initialLoadDelay: 2,

				weatherProvider: "weatherbit",
				weatherEndpoint:"/forecast/daily",  // < --- leave this off and module doesn't work , isn't that what type is supposed to FIX..
				type: "forecast",   // <--  but provider doesn't use this 'required' parm..
				lat: 30.4548443,
				lon: -97.6222674,
				apiKey: "",
			}
		},
```

rejas, 2022-08-16:
yeah, seems wrong to not use the "type" param which is used in other modules and specifically mentioned in the docs.

openweathermap has both params too but only uses the type only when weatherEndpoint is not specified.

maybe switch the usage in this module to that of openweathermap?

sdetweil, 2022-08-16:
yeh, i did  local change to use type..  will fix to override that with the actual endpoint..   but how do you tell user specified ? cause its a default value.. in the provider 

```
	defaults: {
		apiBase: "https://api.weatherbit.io/v2.0",
		weatherEndpoint: "/current",
```

this.config.weatherEndpoint != this.defaults.weatherEndpoint

rejas, 2022-08-17:
With my PR the weatherEndpoint is only used if explicitly set in the config, and not per default. That way existing configs still function (if at all) but new users can rely on the type value to work.

rejas, 2022-08-17:
Looking at the code of other providers it seems that at least darksky should have the same problem...

khassel, 2022-08-17:
> Looking at the code of other providers it seems that at least darksky should have the same problem...

I think we shouldn't investigate in this because "The Dark Sky API and website will continue to function until March 31st, 2023".

Source: https://blog.darksky.net/

sdetweil, 2022-08-18:
@khassel I think u meant invest (spend energy).

it's a tiny thing, for consistency sake

khassel, 2022-08-18:
thanks, yes ...

rejas, 2022-08-18:
It might be tiny, but I still would like to test it before commiting any change. So without any darksky credentials I wont be able to do it.

sdetweil, 2022-08-18:
send an email to my userid at gmail and I will send my key

rejas, 2022-08-18:
Thx for the key, fixed the metric stuff in my WeatherUnitCleanup-PR but somehow darksky seems ok with the "normal" config (type), can you confirm?

sdetweil, 2022-08-19:
yes, darksky worked with type and weatherProvider no other settings
both types worked..

PR #2900 (2022-08-19): Use type value in weatherbit provider
... which is the documented way of configuring a provider instead of relying on undocumented weatherEndpoint param.

Fixes #2899