MagicMirrorOrg/MagicMirror #2901: MMM-NetworkConnection - unable to find server

droidus, 2022-08-18:
I am running MM on a rpi 4b.  Here is the error message that I am experiencing:

`[18.08.2022 06:00.12.338] [ERROR] Whoops! There was an uncaught exception...
[18.08.2022 06:00.12.341] [ERROR] Error: No server found, verify your proxy/network
    at gotServers (/home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/speedtest-net/index.js:621:25)
    at /home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/speedtest-net/index.js:107:16
    at /home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/speedtest-net/index.js:301:7
    at Parser.<anonymous> (/home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/xml2js/lib/parser.js:303:18)
    at Parser.emit (node:events:527:28)
    at Parser.exports.Parser.Parser.parseString (/home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/xml2js/lib/parser.js:313:16)
    at Parser.parseString (/home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/xml2js/lib/parser.js:5:59)
    at exports.parseString (/home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/xml2js/lib/parser.js:354:19)
    at /home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/speedtest-net/index.js:299:5
    at /home/<user>/MagicMirror/modules/MMM-NetworkConnection/node_modules/speedtest-net/index.js:107:16
[18.08.2022 06:00.12.342] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[18.08.2022 06:00.12.343] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues`

DNS resolution is working.  I received this error message several times.

sdetweil, 2022-08-18:
ok, I was able to recreate this. it is a bug in the module  when  the mm config
address:'0.0.0.0'

is set...

not a MagicMirror problem

apparently the speed-test.net library needs a real IP address 

it works ok if address:'localhost' is set

sdetweil, 2022-08-18:
but sometimes it works ok....

droidus, 2022-08-19:
Where is this value defined?

sdetweil, 2022-08-19:
at the top of config.js

khassel, 2022-10-02:
@droidus @MichMich should be closed, no update for more than a month and unrelated to core mm.

closed 2022-10-03 by MichMich