MagicMirrorOrg/MagicMirror #2903: Electron 20 causes rebuild failures in epoll (used by all the GPIO modules, pir, etc)

sdetweil, 2022-09-04:
to recreate 
install MM
switch to develop
install MMM-PIR-Sensor

boom.. 

I opened this issue on epoll, which needs to drop nan and use node-addon-api
but that means a near total rewrite.. SO.. its unlikely any time soon

https://github.com/fivdi/epoll/issues/47

we cannot go above electron 19 at this time

sdetweil, 2022-09-04:
I see that MMM-PIR-Sensor-List uses python to get signals..  and installs ok as it doesn't use epoll

the major PIR modules all appear archived..

bugsounet, 2022-09-04:
Hi, it's related to [this MM topic](https://forum.magicmirror.builders/topic/17174/electron-v20-will-coming-with-next-release-but)

in my case not working with V20.x:

[EXT-Pir](https://github.com/bugsounet/EXT-Pir): use epool library
[EXT-Detector](https://github.com/bugsounet/EXT-Detector): use node snowboy library
[Gateway](https://github.com/bugsounet/Gateway): use node-pty
[EXT-MusicPlayer](https://github.com/bugsounet/EXT-MusicPlayer/) : use node usb

sdetweil, 2022-09-04:
snowboy uses nan

latest node usb should be ok, removed nan

node-pty uses nan

epoll uses nan

bugsounet, 2022-09-06:
Perfect thx @khassel ;)

khassel, 2022-12-02:
any progress here?

Because:

End of Support for 19.x.y

Electron 19.x.y has reached end-of-support as per the project's [support policy](https://www.electronjs.org/docs/latest/tutorial/electron-timelines#version-support-policy). Developers and applications are encouraged to upgrade to a newer version of Electron.

Quoted from https://releases.electronjs.org/releases/stable

bugsounet, 2022-12-03:
If for you it's really important... change it!
My opinion does not count, I am not an official MagicMirror developer (or modules developer)

But do not warn at the last moment!
I think that an special annonce in forum is needed

@bugsounet

bugsounet, 2022-12-03:
Infos:

Tested with electron v22.0.0

`✔ MagicMirror Rebuild Complete`

So should be ok

khassel, 2022-12-03:
Thanks!

So I will update to electron v22 before next release.

rejas, 2022-12-04:
But is the original issue solved (electron rebuild failure) with electron v22?

And will the electron22 requirements be stricter again (fewer architectures supported for instance)?

sdetweil, 2022-12-04:
he has his own rebuild.  we should ship electron-rebuild. 

a secondary problem is that it needs to be installed in the mm node_modules folder. many apps install it locally in their module folder  but that doesn't work anymore.

bugsounet, 2022-12-04:
> he has his own rebuild. we should ship electron-rebuild.

it's +/- the same but simplified for MM²

> a secondary problem is that it needs to be installed in the mm node_modules

I never do this

> many apps install it locally in their module folder but that doesn't work anymore.

with electron-rebuild:
~/MagicMirror/modules/EXT-Pir$ ./node_modules/.bin/electron-rebuild 
⠧ Building module: epoll, Completed: 0make: Entering directory '/home/bugsounet/MagicMirror/modules/EXT-Pir/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
⠴ Building module: epoll, Completed: 0  SOLINK_MODULE(target) Release/obj.target/epoll.node
⠦ Building module: epoll, Completed: 0  COPY Release/epoll.node
make: Leaving directory '/home/bugsounet/MagicMirror/modules/EXT-Pir/node_modules/epoll/build'
✔ Rebuild Complete

```
~/MagicMirror$ cat package.json | grep '"electron":'
                "electron": "^22.0.0"
```
```

sdetweil, 2022-12-04:
@bugsounet  that is because electron-rebuild is installed in the module folder.. 

if u install it in MM folder and then reference there from module folder it works
```
../../node_modules/.bin/electron-rebuild 
```
I have fixed a bunch of modules by making a postinstall script that does that..  vs hard coding electron-rebuild in package.json

bugsounet, 2022-12-04:
Are you really sure that install electron-rebuild in MM folder is a good choice ?
I'm really not sure.
You will make issue for updating MM²
Most of users are lost with errors  on updating

But it's your choice, make as you want ;)
Developers have to done the better to catch any errors (and don't have a black screen)

Best way is installing electron rebuild in dependency of the module (make Electron Rebuild)
on launch of MM² catch any error when loading the library and display the result (bypass black screen)
sample:

[image]

But It's not the subject !

`I am not a reference, My opinion does not count, I am not an official MagicMirror developer (or 3rd  modules developer)`

I leave you to make this decision alone
Happy coding!
@bugsounet

khassel, 2022-12-04:
> But is the original issue solved (electron rebuild failure) with electron v22?

it is "solved" in the same way as with current v19, you have to use the rebuild of bugsounet.

> And will the electron22 requirements be stricter again (fewer architectures supported for instance)?

what do you mean? There are 78 different download options on the release site as for other versions before.

It will be stricter concerning the browser because it contains a newer browser version.

I would like to test it on the `develop` branch now and if there are problems we can revert back to v19 before next release.

rejas, 2022-12-10:
> what do you mean? There are 78 different download options on the release site as for other versions before.

I just remember vaguely that we were quite conservastive with upgrdaing electron in the past. Was that becuase of the spectron dependency or did that have other reasons too?

khassel, 2022-12-10:
from the change log:

```
mm 2.20 electron 19
mm 2.18 electron 16
mm 2.17.1 electron hotfix 13.5.1
mm 2.17 electron 13
mm 2.15 electron 11
before electron 3
```

So I think after upgrading to electron v11 we were not so conservative. We had luck with mm 2.17 because we were using a supported electron version because of the certificate bug (hotfix release 2.17.1).

Thats one reason why I'm not a friend of using an outdated electron version (which gets no updates anymore).

sdetweil, 2022-12-10:
in my opinion we don't need to be bleeding edge. we only have dhtml API calls and generally simple web content, one page, normal css.  (altho I would like to see has() for css. )

iframes and such.

sdetweil, 2023-01-03:
22 does not cause this problem

closed 2023-01-03 by sdetweil