MagicMirrorOrg/MagicMirror #2907: New Install script with separate control server

VMI1994, 2022-09-07:
Hey Michael. I know you already reference a single command install script for MagicMirror2 but I have created a new one which includes a separate control server on port 5000 to easily turn the mirror on and off as well as check the status, system info(using uptime and neofetch), and reboot/shutdown/update-upgrade functions. If you are interested you can check it out at: https://github.com/VMI1994/magic-mirror-setup. Have a great day and thanks for your open source projects.

sdetweil, 2022-09-07:
best if you introduce that to the users and see what they think..

VMI1994, 2022-09-07:
Well please give it a try and let me know what you think(Even in a VM you trash right after).  Also, how do I introduce it to the users?(I'm kinda new at this!)

sdetweil, 2022-09-07:
post a topic, here is my new thing.

for me personally there is not much value in on/off/reboot....

VMI1994, 2022-09-07:
Thanks for the feedback. Where can I find the ‘topic’ section to post?  Also, is it ok if I apply some logging based on what I learn from your scripts(haven’t looked yet but plan to try to digest your code-self teaching here)?

sdetweil, 2022-09-07:
either general or troubleshooting should get attention.

sure, on my scripts, if they can help, use whatever.  ask questions if u need more info.  

pretty easy tho

get env info

check on node and npm

get env setup, install/upgrade whatever

clone
npm install

fixup per env.  pi0w/armv6l

touch up.. css file, etc

then add on auto start
and turn off screen saver

these two sections made standalone if someone changes their mind

upgrade. don't change anything unless requested

test, identify and communicate possible issues.
code that would be overwritten...
then check all modules that should be refresh npm install 

do all on arg apply

IF already on current release, skip apply unless force requested.  it the fastest way to fix all the missing libs(request, valid-url...)

log everything so I can debug and fix

VMI1994, 2022-09-08:
Awesome, thanks for the input. I’ll close this issue now.

closed 2022-09-08 by VMI1994