MagicMirrorOrg/MagicMirror #2909: incorporate remote njk template file configuration for weather

Kayakbabe, 2022-09-11:
I want to do a pull request so I can modify the weather module.
By changing it slightly, it will not affect users if they update, they will not have to edit their configs.. existing configs and future updates to MM will continue to work with my modification.

Now that templating is a real thing for MagicMirror,  I think that having a config variable for a custom weather template would make updating less stressful and easier for those who have modified their weather. Custom templates can have different names than the default current.njk and forcast.njk and could even be placed in the config folder if a path is given for the template file.

To that end I have modified my local weather module instance as follows.

In the weather module weather.js defaults section I added

	defaults: {
		remoteTemplate: "current.njk",

and I altered the getTemplate function to the following:

	// Select the template depending on the display type.
	getTemplate: function () {
		
		if (this.config.remoteTemplate !="") {return this.config.remoteTemplate}
		 else {
		
			switch (this.config.type.toLowerCase()) {
				case "current":
					return "current.njk";
				case "hourly":
					return "hourly.njk";
				case "daily":
				case "forecast":
					return "forecast.njk";
				//Make the invalid values use the "Loading..." from forecast
				default:
					return "forecast.njk";
			}
	   }
	},

If the config value is not placed in config.js, the module will not fail, it will get the template as it does without the modification.

However, this modification allows using a custom template that will not be overwritten by updates.

Benefit of this mod is being able to change the css and html layout without having to create a new module. A user can put their own template in the same folder as current.njk and forecast.njk or as I did they can use a relative or absolute path in the config.

my usage of this mode.
I created mycurrent.njk and myforcast.njk and put them in the mm config folder. 

Then,
in the weather section of my config.js
I added the relative path to my custom template file.

	config: {
		remoteTemplate: "../../../../config/kellyforecast.njk",
	

**MagicMirror² Version**: 2.20.0

rejas, 2022-09-14:
Hi @Kayakbabe this sounds like an interesting and helpful thing. Do you have the PR already in progress or need some help?

rejas, 2022-09-17:
by the way: is that a duplicate of https://github.com/MichMich/MagicMirror/issues/2880 ? can one be closed

rejas, 2022-09-20:
Yes, in the long run this might be interesting for other modules as well. 

But I would propose to first do it in this here since @Kayakbabe seems to be invested in this module. That would not bloat the scope of a potential PR and allow to see how it might be forged into a more general approach later.

khassel, 2024-06-24:
to be honest, I forgot this issue but meanwhile I implemented https://gitlab.com/khassel/MMM-WeatherBridge because I do not like the default weather styling.

@rejas if you find some spare time you can take a look, maybe we can move parts of it into the default weather module.

brilliant0111, 2025-01-12:
> I want to do a pull request so I can modify the weather module.
>
> By changing it slightly, it will not affect users if they update, they will not have to edit their configs.. existing configs and [...]

sdetweil, 2025-01-12:
i think a better approach is to add 3 variables for the template file names, defaulting  as they are now

in the function that returns the hard coded name just return the appropriate template config property

then the user can override any/all of them, no custom code