MagicMirrorOrg/MagicMirror #2910: Weather doesn't show anything anymore

DigiEdAlmere, 2022-09-12:
I don't get it. Suddenly last week the default Weather module doesn't show anything anymore. Not even the header. I created a new ApiKey, but still no data coming through. No error or whatsoever. Is Openweather not functioning anymore?

rejas, 2022-09-12:
works for me without any problem ATM. whats your config (minus the API key)? are there any errors?

Kayakbabe, 2022-09-13:
works for me too. I'm using openWeatherMap as my provider. Which are you using?

DigiEdAlmere, 2022-09-13:
Hi,

This is that part of the config, which has always worked fine:

                {
                        module: "weather",
                        position: "top_right",
                        header: "Het weer vandaag in Almere",
                        config: {
                                colored: true,
                                fade: false,
                                type: "current",
                                location: "",
                                locationID: "2759879", 
                                apiKey: “****"
                        }
                },
                {
                        module: "weather",
                        position: "top_right", 
                        header: "Weersvoorspelling",
                        config: {
                                colored: true,
                                fade: false,
                                type: "forecast",      
                                location: "",
                                locationID: "2759879", 
                                apiKey: “***" 
                        }
                },

Strange huh?

I use OpenWeatherMap

rejas, 2022-09-13:
Using your config with my api key works for me.

The only thing weird in your config is the “ in the apiKey-values. This should be a normal " (like at the end of those two lines

DigiEdAlmere, 2022-09-13:
Strange huh? How does your Weather part in the config looks like?

When I start the mirror with npm start run, I do see this:

[13.09.2022 22:34.39.484] [LOG]   No helper found for module: weather.
[13.09.2022 22:34.39.822] [LOG]   Initializing new module helper …

But it runs without issues, except for weather not showing.

rejas, 2022-09-13:
Maybe try a "npm ci" (clean install) in the mm directory?

DigiEdAlmere, 2022-09-13:
No change.  It almost looks like the whole weather module is blanked out. Like it isn’t in the config at all.

rejas, 2022-09-13:
Do you see any errors in the brwoser console?

sdetweil, 2022-09-13:
ctrl-shift-i ii
on the mm web page  and select the console tab, then enter wea  in the filter field to limit messages to that module

DigiEdAlmere, 2022-09-15:
Nothing. Tried everything. Ending up re-installing it all from scratch. Again. This updating/upgrading of MagicMirror has always been a hassle for me. And every time I end up re-install the whole thing again.

closed 2022-09-15 by DigiEdAlmere

sdetweil, 2022-09-15:
have you tried using my upgrade script?

see
https://github.com/sdetweil/MagicMirror_scripts

DigiEdAlmere, 2022-09-15:
I'll remember that. Finally in a last, last attempt I ran this: sudo apt-get update && sudo apt-get dist-upgrade npm && sudo dist-upgrade pm2

It fixed everything...

sdetweil, 2022-09-15:
the script updates the base, sudo apt-get update,  node & npm to (specific version) , but not pm2..  maybe something to add
I do NOT force OS upgrade