MagicMirrorOrg/MagicMirror #2911: E2E tests never fail

rejas, 2022-09-17:
When working on my weather refactoring I noticed that the E2E tests never fail.

As can be seen after this commit to the branch https://github.com/MichMich/MagicMirror/pull/2849/commits/d4ef2d157e3414ec60e0b9cdc10068ce3705176d

the test suite should obviously fail, but the checks are all green. 

Not sure yet when this bug got introduced, maybe the e2e refactoring has something to do with it?  @khassel do you have an idea?

khassel, 2022-09-17:
> do you have an idea?

not at the moment, will look into this ...

khassel, 2022-10-01:
@rejas can this be closed? Thanks.

rejas, 2022-10-02:
Yeah, thx. Looking into the tests still since that whole done-calls are looking somehow funky...

closed 2022-10-02 by rejas