MagicMirrorOrg/MagicMirror #2913: calendar recurring events are broken since update to node-ical v0.15.x

khassel, 2022-09-20:
Because of #2911 this was not detected earlier.

We upgraded node-ical from `v0.13.0` to `v0.15.1` with mm release `v2.19.0`.

With the newer node-ical version the recurring event tests are not working anymore, I disabled them until this is solved.

Loading `tests/configs/data/calendar_test_recurring.ics` fails with

```bash
[19.09.2022 21:08.00.279] [ERROR] Calendar Error. Could not fetch calendar:  http://localhost:8080/tests/configs/data/calendar_test_recurring.ics TypeError: curr.start.getTime is not a function
at Object.originalEnd (/opt/magic_mirror/node_modules/node-ical/ical.js:423:44)
at Object.END (/opt/magic_mirror/node_modules/node-ical/ical.js:602:26)
at Object.handleObject (/opt/magic_mirror/node_modules/node-ical/ical.js:634:39)
at Object.parseLines (/opt/magic_mirror/node_modules/node-ical/ical.js:686:18)
at Object.parseICS (/opt/magic_mirror/node_modules/node-ical/ical.js:722:18)
at sync.parseICS (/opt/magic_mirror/node_modules/node-ical/node-ical.js:198:15)
at autodetect.parseICS (/opt/magic_mirror/node_modules/node-ical/node-ical.js:229:17)
at /opt/magic_mirror/modules/default/calendar/calendarfetcher.js:72:18
at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
```

I'm to far away from the calendar stuff to solve this at the moment, may @sdetweil can take a look? Thanks.

Other solution could be a downgrade to node-ical `v0.14.1` which is the latest version working.

sdetweil, 2022-09-21:
well, the REAL problem  on the testcase failing is garbage in the testcase in the timezone section, which we don't use, nor ical 

this 
```
TZOFFSETTO:+0000
TZNAME:GMT
DTSTART:19700101T00000--äüüßßß-0   <----- garbage
END:STANDARD
END:VTIMEZONE
```
should be changed to this 
```
TZOFFSETTO:+0000
TZNAME:GMT
DTSTART:19700101T00000
END:STANDARD
END:VTIMEZONE
```

khassel, 2022-09-21:
Tests are fine now, closing this.

closed 2022-09-21 by khassel