MagicMirrorOrg/MagicMirror #2916: squash pull request commits by default

khassel, 2022-09-26:
This is  a proposal to get a cleaner commit history in the future.

So far all commits of a PR are merged in the `develop` branch. E.g. if my PR contains 20 commits (all my tries until I got the right result) they will all remain in the history of this git repo.

I think we should squash these commits, so only 1 commit (which contains all stuff of the e.g. 20 commits delivered) is merged for every PR. This will result in a cleaner history.

To achieve this the settings of this project must be changed, see .e.g. these links:
- https://egghead.io/lessons/github-automatically-squash-commits-on-github-when-merging-pull-requests
- https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/configuring-pull-request-merges/configuring-commit-squashing-for-pull-requests

So this is as usual @MichMich decision ...

sdetweil, 2022-09-26:
oh, this sounds nice. I know I have many pr with multiple commits..

rejas, 2022-09-27:
I am also in favor of a cleaner history, but that would also require from us to have more "focused" PRs with only the necessary changes without other cleanups / fixes etc. Otherwise finding bugs later will be more diffuclt.

MichMich, 2022-09-27:
Agree with @rejas, for most of the PR's this might make thing more complicated. But I'm all for clarity, so let me know if I need to change something.

rejas, 2022-09-27:
You can always decide for each PR if you want to merge it normally or squash it. So we are not fixated on squashing PRs every time. But we should encourage developers to keep their PR simple and clean

sdetweil, 2022-09-27:
>keep their PR simple and clean

agree, but how do I do that without saving the files, dumping my branch  and starting over. hoping I didn't miss anything( which caused the problem to begin w, CHANGELOG, formatting, ...)

rejas, 2022-09-27:
With simple and clean I meant that the PR has one specific purpose: 
- Fix issue X
- Add feature Y

That purpose should be clear when creating a branch on your local fork. 

Of course, it's tempting to fix/improve other stuff on the way to a final PR but one should be wary of that and look if that can be done in a different PR.

sdetweil, 2022-09-27:
I understood. but I submitted the pr w that info

but forgot the CHANGELOG, and the formatting ( don't usually install the dev-dependencies )

khassel, 2022-09-27:
> I am also in favor of a cleaner history, but that would also require from us to have more "focused" PRs with only the necessary changes without other cleanups / fixes etc. Otherwise finding bugs later will be more diffuclt.

I agree (in general) to have more "focused" PRs. But why do you think "mixed" PRs makes finding bugs more difficult? This depends on the commit message of the squashed commit (see project settings later in this post) ...

> You can always decide for each PR if you want to merge it normally or squash it. So we are not fixated on squashing PRs every time.

Project settings

[image]

I would prefer the above settings. With this squashing is default when pressing the Merge-Button, other 2 options are available, you can see this in the short 30 sec. Video [here](https://egghead.io/lessons/github-automatically-squash-commits-on-github-when-merging-pull-requests).

We can control title/description of the PR (which will become the commit message) by reviewing the PR's and not approving them and tell the contributor to change them if needed.

We had similar discussions at work, meanwhile we are squashing every PR, no one wants back to non-squashing ...

khassel, 2022-09-28:
Next proposal:

@MichMich will change the project settings to the above properties **after the next release**. We can test with upcoming new PR's if it works and he can change back if not.

sdetweil, 2022-09-28:
Yes!

rejas, 2022-09-29:
Agreed!

khassel, 2022-10-01:
@MichMich have you changed the above settings?

MichMich, 2022-10-02:
Done!

khassel, 2022-10-04:
Thanks, squashing is working now, here a screenshot how it looks after 3 PR's:

[image]

closed 2022-10-04 by khassel