MagicMirrorOrg/MagicMirror #2922: Version 2.21; Electron update issue (Mirror launches with white screen after upgrade)

nicklam110, 2022-10-02:
Hello all,
I am trying to update my MagicMirror2 software to the latest version, "2.21" upon doing so, I launch the software after the update and am presented with a white screen. Afterwards, I went into the terminal and ran  "npm audit" to see what went wrong during the installation. I have noticed that for some reason, electron was still at version 15.5.4. I ran "npm audit fix --force" to attempt to fix the issue. All this did was place my MagicMirror software into an infinite reboot loop. It boots, the screen is completely white, and it reboots itself to present the same white screen, so on and so forth. I was wondering if anybody else is having the same issue when trying to update the software to the latest version. I ran "git pull && npm install --only=prod --omit=dev" when updating my mirror. Thank you!

sdetweil, 2022-10-02:
please don't EVER  run audit fix..

run my upgrade script w force as the parameter instead of apply

see
https://github.com/sdetweil/MagicMirror_scripts

nicklam110, 2022-10-02:
@stefanbols thank you.. Your scripts helped TREMENDOUSLY... For future reference, should I even run "git pull && npm install --only=prod --omit=dev" when upgrading the software when its behind commits? Or use your two upgrade/apply scripts?

sdetweil, 2022-10-02:
there are tons of issues when there are extra modules loaded into the system.

I cannot teach everyone. 

run my script. if it fails I will fix it

it's only one script. try to examine for issues YOU need to correct (like u changed one of our files on purpose for some reason, (took out precip in the weather template))..  if u don't know, then u will lose that.  so the test run checks, but makes no changes.  I assumed you worked week/months to get it this way, don't want to mess it up.

you DO backup , right?

see my backup restore
https://github.com/sdetweil/MagicMirror-backup-restore

oops, wrong user reference in your post

sdetweil, 2022-10-03:
@nicklam110 can u close this now?

closed 2022-10-03 by nicklam110