MagicMirrorOrg/MagicMirror #2925: weathergov missing method fetchWeatherHourly()

dWoolridge, 2022-10-03:
The fetchWeatherHourly() method does not exist in weathergov.js.  I'd prefer to have it functional because weather.gov gives 6.5 days of hourly forecasts.

I've created the fetchWeatherHourly() method to weathergov.js and tested it for functionality.  It works, but the strange thing is that the new code shows "Loading..." until the first 10 minute refresh is reached, then it works.  The fetchCurrentWeather() and fetchWeatherForecast() populate almost immediately.  

These 3 "fetch" methods are called before the config.js information is read.   Weathergov.js works by first retrieving URL's from the weather.gov API for the current, forecast, and hourly data.  The URLs are read inside the setConfig method.  Weathergov.js' first call (on  startup) to all three fetch(es) fails.  The fetchCurrentWeather() and fetchWeatherForecast() are then called again shortly after the config.js info is read.  For some reason fetchWeatherHourly() is not called a second time along with the other 2 fetch(es).

Any idea on why the fetchWeatherHourly isn't called again at startup, but the other 2 fetch(es) are?

---
```
// ADDED TO weathergov.js

        // Overwrite the fetchWeatherHourly method.
        fetchWeatherHourly() {
                if (!this.configURLs) {
                        Log.info("fetchWeatherHourly: fetch wx waiting on config URLs");
                        return;
                }
                this.fetchData(this.forecastHourlyURL)
                        .then((data) => {
                                if (!data) {
                                        // Did not receive usable new data.
                                        // Maybe this needs a better check?
                                        return;
                                }
                                const hourly = this.generateWeatherObjectsFromHourly(data.properties.periods);
                                this.setWeatherHourly(hourly);
                        })
                        .catch(function (request) {
                                Log.error("Could not load data ... ", request);
                        })
                        .finally(() => this.updateAvailable());
        },
```
---
```
// ADDED TO weathergov.js

        generateWeatherObjectsFromHourly(forecasts) {
                // initial variable declaration
                const days = [];

                // variable for date
                let date = "";
                let weather = new WeatherObject(this.config.units, this.config.tempUnits, this.config.windUnits, this.config.useKmh);
                for (const forecast of forecasts) {
                        weather.date = moment(forecast.startTime.slice(0,19));
                        weather.windSpeed = forecast.windSpeed;
                        weather.windDirection = forecast.windDirection;
                        weather.temperature = forecast.temperature;
                        weather.tempUnits = forecast.temperatureUnit;
                        // use the forecast isDayTime attribute to help build the weatherType label
                        weather.weatherType = this.convertWeatherType(forecast.shortForecast, forecast.isDaytime);

                        // push weather information to days array
                        days.push(weather);

                        // create new weather-object
                        weather = new WeatherObject(this.config.units, this.config.tempUnits, this.config.windUnits, this.config.useKmh);
                }

                // push weather information to days array
                days.push(weather);
                return days;
        },
```

rejas, 2022-10-03:
Hi @dWoolridge it might be easier to check your code if you'd create a PR with it and mark it as a draft. That way I can check it out directly and maybe help you sicne I was working quit a lot in the weather module the last feek weeks :)

dWoolridge, 2022-10-03:
Pull request created:
https://github.com/MichMich/MagicMirror/pull/2929#issue-1395153967

Sorry, but I must have missed the Draft option when I created it.