MagicMirrorOrg/MagicMirror #2926: weathergov Documentation Update

dWoolridge, 2022-10-03:
After a deep-dive into the weathergov.js weather provider script, I believe the documentation needs to be updated...

https://docs.magicmirror.builders/modules/weather.html#configuration-options

**Under "Weather.gov options"**
- apiBase "Possible value" should be 'https://api.weather.gov' as "/points/" is appended in the weathergov.js code.
- weathergov.js no longer uses the **weatherEndpoint** configuration setting as the docs suggest.  This option should be deleted from the documentation.

dWoolridge, 2022-10-03:
Hi rejas,

Please bear with me a bit as I am new to GitHub.  I'm learning a lot as I go along--in addition to having done the GitHub basic training.

The docs that need updated are at docs.magicmirror.builders.  How would I do a GitHub Pull Request for that?

apiBase:  I could go either way on this.  The URL that's ultimately built is:
- https://api.weather.gov/points/{lat},{lon}

I'd look for your recommendation on whether to:
1. Keep the docs for apiBase (possible value) = "https://api.weather.gov/points/" and update the code -OR-
2. Update the docs to apiBase (possible value) = "https://api.weather.gov" and not change the code

dWoolridge, 2022-10-03:
Sorry,  

I just noticed the pull request to change the code for the apiBase.  I'm fine with doing option #1 from my previous post.

...and there is the mention right beneath your post.  I'll have to start reading things with my eyes fully open.

khassel, 2022-10-03:
> The docs that need updated are at docs.magicmirror.builders. How would I do a GitHub Pull Request for that?

the docs are in [this github repo](https://github.com/MichMich/MagicMirror-Documentation), so you should create PR's concerning documentation there, thanks!

dWoolridge, 2022-10-03:
I just realized that changing the code will break the weathergov provider for anyone using it.  

They'd have to change the apiBase definition in the config.js to get it working again.

dWoolridge, 2022-10-06:
Closing Issue.

Code change to remove points are in process in [PR #2927](https://github.com/MichMich/MagicMirror/pull/2927)
Doc changes are in process MichMich/MagicMirror-Documentation [PR 132](https://github.com/MichMich/MagicMirror-Documentation/pull/132)

closed 2022-10-06 by dWoolridge