MagicMirrorOrg/MagicMirror #2942: Improve mocking dates/times in e2e tests

rejas, 2022-10-11:
Some module e2e tests need to fake a certain date/time to be run, especially the calendar module.

There isnt (yet) a general way to mock the current date/time in the e2e tests. Only hackish solutions exist, like for the compliments module for displaying a compliment on a specific date:

https://github.com/MichMich/MagicMirror/blob/a328ce537faf8d9b90d7eb2c09237fe83ae3a827/tests/e2e/modules/compliments_spec.js#L83

Here a mockDate is in the test config which gets read in the mm-js-logic. Using test logic in normal code is of course a no-no!

There are quite a few solutions out there, like sinon, mockDate or adding a jest.mock. Alas, the author hasnt gotten any of them to run for faking a test :-(

Adding the functionality to mock a certain date/time would open up a lot of new test posibilities, for example [this new feature](https://github.com/MichMich/MagicMirror/pull/2939) could be tested very easily.

rejas, 2022-10-11:
Looks like what we'd need, did you try to replace the mockDate construct in the mentioned compliments-test?

> Need we more?

One gotta look through the code where it makes sense to test and fake dates/time to be sure :-)

khassel, 2022-10-11:
sounds like I should assign this to myself ;)

khassel, 2022-10-11:
are these your points in compliments tests?

- `mockDate` is used in `compliments.js`, very ugly, so this should be removed
- some tests run only within specific hours, should run every time and hour set as mock before

khassel, 2022-10-12:
the above mentioned `jest.setSystemTime` works for normal tests (`unit`) but not in our `e2e` test setup because the date/time of the browser has to be mocked. Found no solution for this.

Other possibility is to move such tests back to `electron` section because `playwright` can fake browser date/time.

Here a first (dirty) working example:

```js
const { _electron: electron } = require("playwright");

let electronApp = null;
process.env.MM_CONFIG_FILE = "tests/configs/modules/compliments/compliments_parts_day.js";
process.env.TZ = "GMT";
jest.retryTimes(3);

describe("Compliments", () => {
	beforeEach(async () => {
		electronApp = await electron.launch({ args: ["js/electron.js"] });
	});

	afterEach(async () => {
		await electronApp.close();
	});

	it("Morning compliments for that part of day", async () => {
		expect(await electronApp.windows().length).toBe(1);
		const page = await electronApp.firstWindow();
		expect(await page.title()).toBe("MagicMirror²");
		expect(await page.isVisible("body")).toBe(true);
		page.evaluate(() => {
			Date.now = () => {
				return new Date("01 Oct 2022 10:00:00 GMT");
			};
		});
		let elem = page.locator(".compliments");
		await elem.waitFor();
		expect(elem).not.toBe(null);
		elem = page.locator(".module-content");
		await elem.waitFor();
		expect(elem).not.toBe(null);
		expect(["Hi", "Good Morning", "Morning test"]).toContain(await elem.textContent());
	});
});
```

sdetweil, 2022-10-12:
can;t use set the date/time/tz from bash script between tests
https://www.cyberciti.biz/faq/howto-set-date-time-from-linux-command-prompt/

khassel, 2022-10-12:
may on my machine but I don't know if this would work when running the tests on github (can test it) but also no beautiful approach ...

sdetweil, 2022-10-12:
yes, but the only other thing is to scaffold/proxy the calls made and during test return false info..   testing is SO much fun

rejas, 2022-10-20:
so we can close this now @khassel or do you think we can improve it even further? I am fine with the current state....

khassel, 2022-10-20:
can be closed, if we find more improvements we can do this in a new ticket.

closed 2022-10-20 by khassel