MagicMirrorOrg/MagicMirror #2955: SMHI isn't working with the weather module

el97, 2022-10-20:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 4 4GB 

**Node Version**: 17.8.0 

**MagicMirror² Version**: 2.21.0

**Description**: I am unable to use SMHI in the weather integration. The module just get stuck in "loading" and nothing happens. I tried to use Openweathermap as a provider and that worked without any issues.

**Steps to Reproduce**: Use SMHI as a weather provider. 

**Expected Results**: Current weather when I use SMHI.

**Actual Results**: The module get stuck at loading

**Configuration**: [config.txt](https://github.com/MichMich/MagicMirror/files/9833854/config.txt)

rejas, 2022-10-21:
Thanks for your report. Seems like the smhi api doesnt like lat lon coordinates with more than 6 digits. So please try these config values:

				lat: 59.322665,
				lon: 18.069666,

el97, 2022-10-21:
Thank you for the reply. It didn't make any change sadly.

rejas, 2022-10-22:
Strange. Can you please post your updated config entry for the weather module? Also: can you post any errors from the console?

el97, 2022-10-22:
Here is my config for the weather module. The only related output to the weather module from the console is ` No helper found for module: weather.`

```		{
			module: "weather",
			header: "Väder",
			position: "top_right",
				config: {
					weatherProvider: "SMHI",
					lat: 56.166473,
					lon: 15.229300,
					precipitationValue: 'pmedian',
					type: "current",
						},
		},

rejas, 2022-10-23:
That config entry works for me. 
The output you mention is fron the server part and is fine.
Can you also get the output from the brwoser part (start MM in server mode and brwose with your brwoser to it)?

rejas, 2022-10-23:
Or even easier: Start the MM with 'npm run start:dev' then the browser console appears on the right and you can see error messages etc pp

el97, 2022-10-23:
I managed to find this in the console. 
```
smhi.js:36 Could not load data: this.calculateAT is not a function
(anonymous) @ smhi.js:36
Promise.catch (async)
fetchCurrentWeather @ smhi.js:36
(anonymous) @ weather.js:177

rejas, 2022-10-23:
Ah, yes that got fixed recently with https://github.com/MichMich/MagicMirror/pull/2931

Three possibilites for you:
a) check out the develop branch (if you are familiar with git)
b) wait till 1.1.2023 when the new version with the fix is release
c) manually change the code

sdetweil, 2022-10-23:
use the develop branch

see here how to get it
https://forum.magicmirror.builders/topic/14327/testing-new-fixes-or-solving-current-problems-with-next-release-code

el97, 2022-10-23:
I fixed it! Thanks for the help @rejas

PR #2956 (2022-10-24): Make sure smhi provider api only gets a maximimum of 6 digits coordinates
Fixes #2955
commit dde88601 (2022-10-24): Make sure smhi provider api only gets a maximimum of 6 digits coordinates (#2956)

Fixes #2955
closed 2022-10-23 by el97