MagicMirrorOrg/MagicMirror #2967: Problems with openweather api?

jojo2203, 2022-11-18:
Hello,

are your Openweather api's working . Have problems since about 1 week. When I query the api directly in the browser line I also get an error message. Are there any other providers. What are your experiences? Thank you.

jojo

rejas, 2022-11-18:
I have no problem with the openweather api. What exactly is the error message you get and what version of MagicMirror are you running?

khassel, 2022-11-21:
We had already a discussion in the forums about api changes of openweather.

I looked into their api stuff and onecall v2.5 is not mentioned anymore. The normal endpoints weather and forecast are still on v2.5.

Problem: For onecall 3.0 you need a special api key (as already mentioned here) with a subsciption where you have to give your payment data (1000 request free per day, limit is 2000 per day but you can set your limit to 1000 if you want it for free … ).

So (for default weather module) if you use endpoints weather and forecast and old onecall v2.5 everything works out of the box with old api keys, if someone use new onecall v3.0 he needs an api key for 3.0 and has to set `apiVersion: "3.0"` in the config …

May this is unrelated to the above problem but we should keep this in mind.

sdetweil, 2022-11-21:
thx, I missed the API version change.  

course this will affect other modukes too

jojo2203, 2022-11-22:
Hello,

I've only just registered with Openweathermap (one week) , so I probably have an Api version 3.0. 25 ... 50 calls a day are enough for me. Can I switch back to the old version 2.5? Or where can you limit the number of views? Didn't find it? I always end up with the normal payment options. Thanks.

sdetweil, 2022-11-22:
as @khassel said add config Parm
```
apiVersion:'3.0',
```
to the default weather module config

rejas, 2023-01-15:
Hi @jojo2203 is this still an issue for you? Or can we close this ticket?

jojo2203, 2023-01-17:
You can close it. I have had to stop my work for the time being.

 
 

Gesendet: Sonntag, 15. Januar 2023 um 22:58 Uhr

closed 2023-01-17 by rejas