MagicMirrorOrg/MagicMirror #2969: playwright v1.28 breaks electron tests

khassel, 2022-11-22:
```js
node@46e0815dd8b0:/opt/magic_mirror$ npx jest tests/electron/env_spec.js
 FAIL   electron  tests/electron/env_spec.js
  Electron app environment
    ✕ should open browserwindow (470 ms)
  Development console tests
    ✕ should open browserwindow and dev console (439 ms)

  ● Electron app environment › should open browserwindow

    TypeError: Cannot read properties of null (reading 'locator')

      37 | exports.getElement = async (selector) => {
      38 |      expect(global.page);
    > 39 |      let elem = global.page.locator(selector);
         |                             ^
      40 |      await elem.waitFor();
      41 |      expect(elem).not.toBe(null);
      42 |      return elem;

      at Object.locator [as getElement] (tests/electron/helpers/global-setup.js:39:25)
      at Object.getElement (tests/electron/env_spec.js:13:32)
```

so we should not update at the moment.

khassel, 2022-11-29:
see https://github.com/microsoft/playwright/issues/18928

mxschmitt, 2022-12-05:
That would be a fix for 1.28:

```diff
diff --git a/package.json b/package.json
index b805bb9f..6f3038c9 100644
--- a/package.json
+++ b/package.json
@@ -59,7 +59,7 @@
 		"jsdom": "^20.0.0",
 		"lodash": "^4.17.21",
 		"nyc": "^15.1.0",
-		"playwright": "^1.26.1",
+		"playwright": "^1.28.1",
 		"prettier": "^2.7.1",
 		"pretty-quick": "^3.1.3",
 		"sinon": "^14.0.0",
diff --git a/tests/electron/env_spec.js b/tests/electron/env_spec.js
index f962949c..2523e6f4 100644
--- a/tests/electron/env_spec.js
+++ b/tests/electron/env_spec.js
@@ -1,5 +1,6 @@
 // see https://playwright.dev/docs/api/class-electronapplication
 
+const events = require("events");
 const { _electron: electron } = require("playwright");
 
 let electronApp = null;
@@ -16,6 +17,7 @@ describe("Electron app environment", function () {
 	});
 
 	it("should open browserwindow", async function () {
+		await electronApp.firstWindow();
 		expect(await electronApp.windows().length).toBe(1);
 		const page = await electronApp.firstWindow();
 		expect(await page.title()).toBe("MagicMirror²");
@@ -36,6 +38,7 @@ describe("Development console tests", function () {
 	});
 
 	it("should open browserwindow and dev console", async function () {
+		while (electronApp.windows().length < 2) await events.once(electronApp, "window");
 		const pageArray = await electronApp.windows();
 		expect(pageArray.length).toBe(2);
 		for (const page of pageArray) {

```

khassel, 2022-12-05:
Thank you @mxschmitt for your hints but it still doesn't work.

`await electronApp.firstWindow();` throws 
```
electronApplication.firstWindow: Timeout 30000ms exceeded while waiting for event "window"`
```

So the problem seems to be that no electron window event is created. Same code works if I downgrade to playwright v1.27.1.

khassel, 2022-12-20:
this will be (hopefully) fixed with next playwright version v1.30 (see https://github.com/microsoft/playwright/issues/19412)

mxschmitt, 2022-12-20:
We'll cherry-pick it to 1.29.1

rejas, 2022-12-22:
Fixed

closed 2022-12-22 by rejas