MagicMirrorOrg/MagicMirror #2972: Magicmirror doesnt start in fullscreen after rebooting

stormyark, 2022-11-26:
Magicmirror doesnt start in fullscreen after rebooting

I followed the introduction from [link text](https://docs.magicmirror.builders/configuration/autostart.html#using-pm2) to autorun magicmirror on startup, but after rebooting this is what i see: 

[image]

rejas, 2022-11-27:
Could you upload the image again?

stormyark, 2022-11-27:
I reuploaded the image

sdetweil, 2022-11-27:
I've seen this a number of times. usually electron gets confused from a prior app usage.

so do this, stop MagicMirror 
```
cd ~
rm -rf .config/Electron
```

note the dot in front of config

this will erase the saved Electron info

start MagicMirror

stormyark, 2022-11-27:
How do I stop magicmirror? If I press STRG+Q the magicmirror restarts :((

sdetweil, 2022-11-27:
ctrl-m to minimize, and then on terminal window, pm2 stop all

sdetweil, 2022-11-27:
pm2's job is to keep the app running

stormyark, 2022-11-27:
[image]

sdetweil, 2022-11-27:
that looks like u are using VLC or some windowing system to access them system.  

does it have a real display?

stormyark, 2022-11-27:
I am running a linux mint on virtual box to test magicmirror
[image]

That are my display settings

sdetweil, 2022-11-27:
electron appears to have some trouble getting the display size.     

its not our code, so I don't know what the constraints are ...
you might look thru the Electro github issues. 

https://github.com/electron/electron/issues?q=is%3Aissue+is%3Aopen+fullscreen+  there are 35 open related to fullscreen

what is the size definition?   1024x768?  1920X1080?

stormyark, 2022-11-27:
Ill go through the issues, thank you so much for your help.

My size defenition is 1920x1080
and in the electon.js I changed the width to 1920 and the high to 1080 so that magicmirror at leasts covers the screen

KristjanESPERANTO, 2022-11-27:
@stormyark Just out of curiosity: Why are you using MagicMirror in a VM?

sdetweil, 2022-11-27:
@stormyark you can change those settings thru config.js, without editing any of the source files.

that you were able to solve part of the problem w electron options confirms that electron is confused

stormyark, 2022-11-27:
@KristjanESPERANTO I Just wanted to test it a bit before the raspi arrives

stormyark, 2022-11-27:
@sdetweil could I possibly write a comand in the confog.js to run fullscreen?

sdetweil, 2022-11-27:
@stormyark the default is fullscreen, in kiosk mode (no title bar, etc)

as I said  it's 'fullscreen' that is confused in the vm

khassel, 2023-01-14:
> Now i looks like this :(

I think this is normal behaviour in VirtualBox. MagicMirror is running fullscreen in the VM, but you have still the Windows-VirtualBox-Window around, AFAIK you cannot control this from inside the VM.

You can click on the VM-Window and play around with Ctrl(Right)+C or Ctrl(Right)+F.

This is not an electron or magicmirror problem and on a pi fullscreen will work.

closed 2023-02-12 by khassel