MagicMirrorOrg/MagicMirror #2974: Weather.js module throws error

sprestel, 2022-12-03:
After update to MM v2.21.0 I'm not able to see the weather.

Could be related to this https://forum.magicmirror.builders/topic/17059/default-weather-module-still-not-displaying/31?lang=de

https://github.com/MichMich/MagicMirror/blob/9e0293047f8a8c0ced7d9fd8de1b839c06baf9cb/modules/default/weather/weather.js#L159

[image]

## workaround

I removed these new lines https://github.com/MichMich/MagicMirror/commit/df0f048ecc018d55119108d7bb7682f3ea3f80b0#diff-f72a25530e6dd944b25232a4d10475685f898a8da89055f070912fc38c35f7d4R157-R165

sdetweil, 2022-12-03:
what browser is being used here

sprestel, 2022-12-03:
Current chromium

sdetweil, 2022-12-03:
not electron.. 

just the js engine in the browser is down level, we need es6 level

khassel, 2023-01-13:
> I removed these new lines

removing these lines is a hack which may works for you not for others.

We need more info about your environment (as requested in the issue template) to help you.

closed 2023-02-12 by khassel