MagicMirrorOrg/MagicMirror #2978: Add additional tests to the pre-commit hook

spyder007, 2022-12-15:
Currently, this repository has a pre-commit check (using husky) that runs `npm run lint:staged`

My proposal is to add additional checks (already configured in `packages.json`) to the pre-commit to ensure that the desired tests (including linting tests) are performed prior to a commit.

Based on what is configured today, my suggestion is to run the following as part of pre-commit:
```sh
npm run lint:staged
npm run test:js
npm run test:css
npm run test
```

khassel, 2022-12-15:
At the moment this is already tested when someone creates a PR.

AFAIS with the above changes all the test stuff is done if you make a local commit. Many developers don't make one final commit where everything fits, they do x commits until the feature is ready. I don't think they want to wait every time  ~5 min. until the hook is ready.

I'm doing some stuff on windows without having any deps installed, therefore I did already the `if command -v npm &> /dev/null; then` around the `npm run lint:staged` to be able to push something (see https://github.com/MichMich/MagicMirror/pull/2788).

These changes may prevent unnecessary communication in PR's, but for me it is unnecessary patronizing of the developer.

May others are more open minded to this ...

If the majority is for implementing this there should be an environment variable where developers can disable this if they want.

sdetweil, 2022-12-15:
I'm not a professional developer, and don't know all the things that need to be done to pass submissions.  so any help is great news. can keep from making 19 tiny commits, oops, no contribute file update, oops, no lint.. etcetera

which have nothing to do w the functional change I am making. 

I know we added squash recently to try the clean up some of this.

khassel, 2022-12-15:
> can keep from making 19 tiny commits, oops, no contribute file update, oops, no lint.. etcetera

you dont want to run the (longrunning) tests with every commit

You can run the tests by yourself before creating a PR, if you forget it, the PR will tell you

sdetweil, 2022-12-15:
you are right. I am not interested in running the tests most times.

rejas, 2022-12-15:
Thanks for you proposal @spyder007 but I have to agree with the other two, running the tests during pre-commit is not helping really.

spyder007, 2022-12-16:
I'm clearly a bit behind on the discussion, so I'll add my two cents and close this issue and PR:
1) Husky can configure both `pre-commit` and `pre-push` hooks, meaning it is possible to separate linting in pre-commit and testing pre-push.  
2) Husky allows you to skip these verifications at any time with either `--no-verify` or by setting an [environment variable](https://typicode.github.io/husky/#/?id=bypass-hooks).  I would assume this would address some of the issues above.
3) As a developer new to the project, unless I go digging into the `.github` folder or digging into the `package.json`, it's difficult to find solid instructions for contributions, which should include things like executing tests and linting, how you want code rebased/merged (do you want force-pushes?), etc.  I realize many of these things are part of the workflows you all have, but as a first time contributor, it is something of a guessing game, especially since my first PR doesn't even run the workflows until they are looked at.

That said, I'll close this issue and the associated workflow.  Thanks for the consideration.