MagicMirrorOrg/MagicMirror #2981: Module wather: Openweathermap: current and hourly show wrong temperature

Coernel82, 2022-12-24:
**Platform**
- Firefox 108.0.1 on Windows to display
- electron: how do I find out? I googled it but didn't find any solution how to find out. Maybe 19.1.0 as I found ``"optionalDependencies": {
		"electron": "^19.1.0"`` in the package.json
- Raspberry Pi 3b+
- Raspian OS 32bit (Debian 11.6)
- Raspberry Pi 3 Model B Rev 1.2
**Node Version**
- 16.19.0
**MagicMirror² Version**:
- 2.21.0
**Description**:
Instead of a current temperature the weather module with openweathermap shows 26,6°. Current temperature really is around 7°C.
**Steps to Reproduce**:
Use my config.
**Expected Results**:
Correct temperature (around 7°C)
**Actual Results**:
26.6°C
**Configuration**
```java
 		
let config = { //bla bla bla
				units: "metric",
		tempUnits: "metric",
		windUnits: "metric",
modules :[
			module: "weather",
			position: "top_right",
			config: {
				// using apiVersion: 3.0, results inst "loading..."
				units: "metric",
				tempUnits: "metric",
				windUnits: "metric",
				degreeLabel: true,
				appendLocationNameToHeader: false,
				weatherProvider: "openweathermap",
				type: "current",  // same problem four "hourly"
				location: "Overath",
				locationID: "2855859", 
				apiKey: "mysecretapi",
				weatherEndpoint: '/onecall',
				updateInterval: 6000000,
				animationSpeed: 0, 
				decimalSymbol: ",",
				initialLoadDelay: 400,
			}
		}, 
 		{
			module: "weather",
			position: "top_right",
			config: {
				degreeLabel: true,
				appendLocationNameToHeader: false,
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "Overath",
				locationID: "2855859",
				apiKey: "mysecretapi",
				weatherEndpoint: '/forecast',
				updateInterval: 6000000,
				animationSpeed: 0, 
				decimalSymbol: ",",
			}
		}, 
// bla bla bla
```

sdetweil, 2022-12-24:
from the doc, if /onecall is used as the endpoint 
lat and lng are required and location/locationid will not be used

[image]

Coernel82, 2022-12-24:
Can't believe I missed that though I have been reading the documentation thoroughly!
However may I suggest that the information also is added in "General options --> type"?
The way it is written now is as if in a cooking book there is a recipe for cake containing milk, butter and a pinch of salt. And then 20 pages later there in the glossary it says: sugar: please note that sugar is required for baking a cake.

sdetweil, 2022-12-24:
well, weather is a kitchen sink kind of module, does 99,000 things

the general options apply to all providers
formatting, display 

but then the providers have specific parms., and some are interdependent as you've discovered.

rejas, 2022-12-24:
I am always happy about PRs in the documentation repo that clean things up :-)

Coernel82, 2022-12-24:
I will do so tomorrow, you will get your Pull Request!

Coernel82, 2022-12-24:
OK.... just had a look at this repository and I am stuck. Which repository do I have to fork to make a pull request for https://docs.magicmirror.builders/modules/weather.html

sdetweil, 2022-12-24:
this repo
https://github.com/MichMich/MagicMirror

sdetweil, 2022-12-25:
this repo
https://github.com/MichMich/MagicMirror-Documentation

Coernel82, 2022-12-25:
Ok, and where is the specific file?

rejas, 2022-12-25:
https://github.com/MichMich/MagicMirror-Documentation/blob/master/modules/weather.md

Coernel82, 2022-12-25:
Thank you. Will make a PR tonight or latest by tomorrow.

Coernel82, 2023-01-10:
weather.md updated. Problem solved.

closed 2023-01-10 by Coernel82