MagicMirrorOrg/MagicMirror #2986: MagicMirror won't start - Electron launches automatically to blank screen

dapter22, 2023-01-02:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:
[https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 14 or later (recommended is 16).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

dapter22, 2023-01-02:
I reinstalled MM a few months ago and it was working find. About two weeks ago it closed by itself and I couldn't relaunch it - Electron opens to blank screen. At that point I found a fix and it seemed to work but now the same thing is happening and I don't remember what I did last time. Electron seems to launch automatically and I get "EADDRINUSE" error as it seems something is already using port:8080. When I ctrl+q the Electron window it just launches again. Used lsof -i tcp:8080 to kill process on that port but it just closes the Electron window and then launches it again to blank screen. Stumped on this one. Please help!

dapter22, 2023-01-02:
also for some reason, the web browser will not launch - I'm guessing because the internet connection is being used?

sdetweil, 2023-01-02:
pm2 stop all

sdetweil, 2023-01-02:
pm2's job is to keep mm running. so if u kill it w ctrl-q, pm2 restarts.

you cannot run two instances at once w the same port..

dapter22, 2023-01-02:
I had to reinstall chromium-browser and your command worked - thanks!!!

sdetweil, 2023-01-02:
@dapter22 is this on  a pi0w?

sdetweil, 2023-01-04:
@dapter22 can we close this?

closed 2023-01-13 by khassel