MagicMirrorOrg/MagicMirror #2987: Weather module using Environment Canada, days are reversed.. but not weather information!

etrefal, 2023-01-02:
Platform: Raspberry Pi 3 on Raspbian Buster
Node version: v16.19.0
MagicMirror version: 2.22.0

Description: I use the weather module with Environment Canada as provider.  It always worked well, until today when I noticed that the day names are reversed.  The weather information is *not* reversed though.  I have attached two screenshots - one is the MagicMirror display, the other is the Environment Canada webpage.  Note that these are in french.  The day names displayed on the MagicMirror are "Aujourd'hui" (Today) - which is a monday), "Demain" (Tomorrow), then "sam." (Saturday), "ven." (Friday), "jeu." (Thursday) and then "mer." (Wednesday).  These should be "Aujourd'hui, Demain, mer., jeu., ven. and sam."

Note that the weather information is correct.  The weather forecast is in the correct order when you compare to the Environment Canada screenshot. 

I did upgrade MagicMirror to the latest version yesterday afternoon.  But I only noticed the error a few minutes ago.

Configuration: My edited config.js file is also attached.

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.
[image]
[image]

rejas, 2023-01-04:
Thx for the report @etrefal I can reproduce this, will have a fix on the weekend ready

etrefal, 2023-01-07:
I had a notification of a new commit, so I did a **git pull && npm install --only=prod --omit=dev**

Weather forecast with Environment Canada is still not fixed... today is Samedi (Saturday), tomorrow is Dimanche (Sunday).   So Sunday appears twice - as Tomorrow (Demain) and as Sunday (Dim.)  Please see attached screenshot.
[image]
The weather forecast is correct, and in the correct order.  Only the day names are wrong.  In the screenshot above, they should be:
Aujourd'hui (Today)
Demain (Tomorrow)
lun. (Monday)
mar. (Tuesday)
mer. (Wednesday)
jeu. (Thursday)

khassel, 2023-01-07:
we collect all fixes on the `develop` branch, releases are only quaterly so next on April 1.

You can use the `develop` branch so far or wait until next release or do the one line fix from the PR in your code directly (in last case don't forget to revert before next git pull)

etrefal, 2023-01-07:
Understood.  I edited the **envcanada.js** file to reflect the change, and will update again in April.

Thanks!

sdetweil, 2023-01-07:
@etrefal if you used my upgrade script, it would detected that you were using the develop branch and would have switched u back to master before the version upgrade

etrefal, 2023-01-08:
@sdetweil - I didn't know about your upgrade script.  I downloaded it on the Pi a few minutes ago.  I guess since I edited the envcanada.js file manually it might not help right now, but it might in April!  I'll keep that in mind when we get there, thanks!

sdetweil, 2023-01-08:
u don't have to download, just copy/paste from the repo readme online

rejas, 2023-01-08:
> Understood. I edited the **envcanada.js** file to reflect the change, and will update again in April.
> 
> Thanks!

Good to hear that it fixed your problem for real. Can we close this issue now?

khassel, 2023-01-08:
> Can we close this issue now?

I thought we close such issues when the fix/feature is released, see Collaboration.md (?)

sdetweil, 2023-01-08:
I think and confirmed by the op

etrefal, 2023-01-08:
Yes, the issue could be closed now, the fix works!

closed 2023-01-08 by etrefal

rejas, 2023-01-08:
> > Can we close this issue now?
> 
> I thought we close such issues when the fix/feature is released, see Collaboration.md (?)

True. Maybe we should add a label "Fix ready in next release" (or something like that) to those issues so that we can see which are "done"

PR #2992 (2023-01-07): Fix wrong day labels in envcanada forecast
Fixes #2987

lastDate isnt a unix date, but just a normal moment-date
commit a975b44f (2023-01-07): Fix wrong day labels in envcanada forecast (#2992)

Fixes #2987

lastDate isnt a unix date, but just a normal moment-date