MagicMirrorOrg/MagicMirror #2988: "Added css class names "today" and "tomorrow" for default calendar"

Haskar, 2023-01-03:
Hi, 

how do I deactivate the colored underlines for "today" and "tomorrow" events in the default calendar? I could not find anything in the documentation. 
The classes are not defined in my calendar.css file and I don't see where the "default" values come from :(

sdetweil, 2023-01-03:
can u post a screen shot of the cal outout?   I don't see an underline

Haskar, 2023-01-03:
Thanks for the friendly reply and the hint. 

".today" and ".tomorrow" was defined in the CSS file of another module I use (MMM-Todoist). I commented it out and it is working fine now!

I should have been able to figure it out myself :(

closed 2023-01-03 by Haskar

sdetweil, 2023-01-03:
so, to fix this just add

.MMM-Todoist and space

in front of those classes to make them relative to that module only