MagicMirrorOrg/MagicMirror #2989: Version 2.22 doesn't run on Raspi OS Stretch,  undefined symbol: gbm_bo_get_modifier

sdetweil, 2023-01-03:
users are reporting that after upgrade to 2.22 they are seeing a new Electron error

```
.node_modules/[…]/dist/electron: symbol lookup error: undefined symbol: gbm_bo_get_modifier.
```

the desktop env is downlevel from what is used by electron V22. (interesting coincidence) 

only fix is to upgrade OS..  I don't think the software upgrade works as next version is buster which has been superceded by bullseye.

rebuild of sd card at bullseye resolves the problem..

KristjanESPERANTO, 2023-01-03:
Do I understand correctly that the problem only occurs with stretch? stretch has reached its end of support. Such an old, no longer supported version should not be used for security reasons, nor should we worry about problems with it.

sdetweil, 2023-01-03:
yes, this was really posted for others to search on. not for the mm team to DO anything

rejas, 2023-06-01:
So we can close it?

sdetweil, 2023-06-01:
can we leave it for one more release?  users don't know how to search closed issues

rejas, 2023-06-01:
maybe worth putting it into the documentation then?

khassel, 2023-07-02:
> can we leave it for one more release?

@sdetweil  so can we close it now?

sdetweil, 2023-07-02:
sure

closed 2023-07-02 by sdetweil