MagicMirrorOrg/MagicMirror #2995: run tests against arm32v7 and/or arm64v8 architectures instead against amd64

khassel, 2023-01-07:
Our tests are running on amd64 architecture at the moment but the most users are running on a raspberry pi 32-bit (`arm32v7`) or 64-bit (`arm64v8`).

Inspired by [this forum thread](https://forum.magicmirror.builders/topic/17513/electron-rebuild-problems-with-magicmirror-2-22-0) concerning problems with `electron-rebuild` I think it would be better to run our tests against arm architectures.

We could also add more tests e.g. `npm install electron/rebuild`.

I tested this already with this yaml file and the tests are running:

```yaml
# This workflow will do a clean install of node dependencies, build the source code and run tests across different versions of node
# For more information see: https://help.github.com/actions/language-and-framework-guides/using-nodejs-with-github-actions

name: "Run Automated Tests"

on:
  push:
    branches: [master, develop]
  pull_request:
    branches: [master, develop]

permissions:
  contents: read

jobs:
  test:
    runs-on: ubuntu-latest
    timeout-minutes: 30
    steps:
      - name: Checkout code
        uses: actions/checkout@v3
      - name: Setup Qemu
        uses: dbhi/qus/action@main
        with:
          targets: arm aarch64
      - name: Get docker image
        uses: docker://arm32v7/node:18-bullseye
      - name: Install dependencies and run tests
        run: |
          Xvfb :99 -screen 0 1024x768x16 &
          export DISPLAY=:99
          npm run install-mm:dev
          touch css/custom.css
          npm run test:prettier
          npm run test:js
          npm run test:css
          npm run test
```

Before I spend more time at this I would appreciate feedback if we want to go this way.

The above yaml file is not complete, it misses the different node versions and different architectures.

Another question is if we really want to run 6 tests instead of 3.

KristjanESPERANTO, 2023-01-08:
Switch architecture(s): Seems like a good move.

6 Tests: You end up with 6 tests because every node version is tested with both architectures, right? So far testing has been done with amd64, maybe just switching to arm64 will suffice.

rejas, 2023-01-08:
Testing with what most users use is a good idea. 
As for how many tests are run: If 32 is used regularly we should test it IMO (I dont think test time is a problem on github)

khassel, 2023-01-08:
Forget the above workflow because it is not running on arm.

I meanwhile managed to get an arm workflow running but the results are sobering. Beside a horrible performance only unit tests are running.

So I close this and may reopen if I find another way ...

closed 2023-01-08 by khassel