MagicMirrorOrg/MagicMirror #2996: Weather module: add incremens for hourly

Coernel82, 2023-01-08:
It would be very great if the default weather module would be able to show the ``hourly`` weather with for instance ``increments = 2`` resulting in:

* 12:00
* 14:00
* 16:00

``increments = 3`` resulting in:
* 12:00
* 15:00
* 18:00

and so on...
as on smaller screens hourly clutters the screen very much.

I'd be very happy about this option!

khassel, 2023-01-08:
@rejas can you may take a look at this?

rejas, 2023-01-08:
already on it, might have something soonish ;-)

Coernel82, 2023-01-08:
Wow, this would be great! I wish I could program more than I am able to at the moment!

rejas, 2023-01-08:
PR created if you want to check it out

Coernel82, 2023-01-08:
Cool! Is there a pull command I can use?
S.th. like ```git pull https://github.com/MichMich/MagicMirror/pull/2998```?

Coernel82, 2023-01-09:
OK... I found something and tried ``git fetch origin pull/2998/head:MichMich/MagicMirror`` - let's see if it worked.

The new variable is: ``hourlyForecastIncrements: 1,`` by the way. Set it to 4 and will report back!

Coernel82, 2023-01-09:
The git fetch command didn't work but finally figured it out:
```
git fetch origin pull/2998/head:MagicMirror
git checkout MagicMirror
```
and: It works so far. I will keep testing for the next days.

Coernel82, 2023-01-10:
still working! Happy that this will be in the next major version in april!

PR #2998 (2023-01-10): Add option to show hourly forecast in increments
Adds new config option to show weather forecast for every X hour (default value is 1 which reflects the current behaviour)
Also adds tests for hourly forecast
Fixes #2996
commit 6e80e5a2 (2023-01-10): Add option to show hourly forecast in increments (#2998)

Adds new config option to show weather forecast for every X hour
(default value is 1 which reflects the current behaviour)
Also adds tests for hourly forecast
Fixes #2996
closed 2023-01-10 by Coernel82