MagicMirrorOrg/MagicMirror #2997: calendar: not showing event location

NimdaBE, 2023-01-08:
Hello,

i'm using the latest version of MM (2.22.0) and the default calender is not showing event locations. I'm using following code in my config file:

module: "calendar",
			position: "middle_center",
			header: "Dieter",
			maximumEntries: "10000",
			wrapEvents: "true",
			showLocation: "true",
			broadcastEvents: "true",
			broadcastPastEvents: "true",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						name: "DVF",
						color: "#5ac1da",
						url: "webcal://calendar.google.com/calendar/ical/redacted/basic.ics"
					}
]
			}

The calendars contain event location information (i checked by downloading the ics file and reading the contents with notepad++). 
- Do i have to change or add parameters to show the event locations?
- Does it matter that i have multiple google calenders defined in my config? Events are showing fine for all calendars.
- No custom modules are installed. This is as vanilla as can be...

best regards,
Dieter

sdetweil, 2023-01-08:
all of these settings 
```
maximumEntries: "10000",
wrapEvents: "true",
showLocation: "true",
broadcastEvents: "true",
broadcastPastEvents: "true",
```
must be INSIDE the

```
config: {
```

NimdaBE, 2023-01-09:
ok, what a stupid mistake to make :-/

You can close this and mark it with a pebkac-tag

closed 2023-01-09 by NimdaBE

sdetweil, 2023-01-09:
computers are SOOOOO picky!