MagicMirrorOrg/MagicMirror #2999: CORS proxy enabled by default in newsfeed module

steepl, 2023-01-09:
I'm not sure if this is correct or not, but after upgrading to version 2.22.0 the CORS proxy for the newsfeed module seems to be enabled by default. I could only turn it off by changing `if (useCorsProxy === undefined) useCorsProxy = true;` to `if (useCorsProxy === undefined) useCorsProxy = false;` inside modules/default/newsfeed/node_helper.js. Shouldn't it be set to false by default?

sdetweil, 2023-01-09:
it should work.

if it does, why do you care?

if it doesn't, then we should fix it

steepl, 2023-01-09:
@sdetweil I noticed the change because I'm using a module (MMM-News-QR) that generates QR codes based on the newsfeed URL. And the module now has to filter the URL to remove the CORS proxy (which seems like bad code to me). It doesn't bother me as I can just change the code as stated above, but I figure more people are going to have the problem after upgrading.

khassel, 2023-01-09:
you can add `useCorsProxy: false` to every feed in the `config.js` where you don't want to use cors.

steepl, 2023-01-09:
@khassel Thanks, I will try that out tomorrow. I added it to the module config directly (not the feeds), but that did not seem to work.

steepl, 2023-01-10:
Closing the issue, adding useCorsProxy to the feeds works.

closed 2023-01-10 by steepl